ARMmbed/mbed-os #7457: Avoid compiling features/cellular files when not needed

adustm, 2018-07-10:
### Description
I compile mbed-os for DISCO_L475VG_IOT01A 
I see undesired files, as this target is not concerned by cellular things
```
Compile [  3.8%]: AT_CellularBase.cpp
Compile [  4.0%]: ATHandler.cpp
Compile [  4.1%]: AT_CellularDevice.cpp
Compile [  4.2%]: AT_CellularInformation.cpp
Compile [  4.3%]: AT_CellularPower.cpp
Compile [  4.4%]: AT_CellularNetwork.cpp
Compile [  4.5%]: AT_CellularSIM.cpp
Compile [  4.7%]: CellularLog.cpp
Compile [  4.8%]: AT_CellularSMS.cpp
Compile [  4.9%]: CellularUtil.cpp
Compile [  5.0%]: AT_CellularStack.cpp
Compile [  5.1%]: QUECTEL_BC95.cpp
Compile [  5.2%]: QUECTEL_BC95_CellularPower.cpp
Compile [  5.4%]: QUECTEL_BC95_CellularSIM.cpp
Compile [  5.5%]: QUECTEL_BC95_CellularNetwork.cpp
Compile [  5.6%]: QUECTEL_BC95_CellularStack.cpp
Compile [  5.7%]: QUECTEL_BG96_CellularPower.cpp
Compile [  5.8%]: QUECTEL_BG96_CellularNetwork.cpp
Compile [  5.9%]: QUECTEL_BG96.cpp
Compile [  6.1%]: QUECTEL_BG96_CellularSIM.cpp
Compile [  6.2%]: QUECTEL_BG96_CellularStack.cpp
Compile [  6.3%]: QUECTEL_UG96_CellularNetwork.cpp
Compile [  6.4%]: QUECTEL_UG96.cpp
Compile [  6.5%]: QUECTEL_UG96_CellularPower.cpp
Compile [  6.6%]: TELIT_HE910_CellularNetwork.cpp
Compile [  6.8%]: TELIT_HE910.cpp
Compile [  6.9%]: TELIT_HE910_CellularPower.cpp
Compile [  7.0%]: UBLOX_PPP.cpp
```
Could someone take the responsibility of checking completely the CELLULAR feature and add the correct `#ifdef `to avoid compiling those files when not needed ?

I think it could be 
`#ifdef CELLULAR_DEVICE`
or
`#ifdef <TARGET_NAME>`

There are maybe other files in this feature
Thanks in advance
Armelle

### Issue request type
    [] Question
    [ ] Enhancement 
    [X] Bug

0xc0170, 2018-07-25:
This is how the build system works. You can use mbedignore file to ignore files, otherwise it builds it all

I'll close this as won't fix (works as designed).

adustm, 2018-07-25:
Hi @0xc0170 @bulislaw 
Please consider this request. 
The build system works with directory names and defines in the code and .json file

If you compile `DISCO_L475VG_IOT01A`, you won't compile files that are located in `targets/TARGET_STM/TARGET_STM32F7` directory

I would assume that if I am not using any define for QUECTEL UBLOX or TELIT cellular in my `mbed_app.json`, I should be able not to compile those files at all. 
**This is already the case for features/cellular/easy_cellular/CellularConnectionFSM.cpp at line 20**
`#ifdef CELLULAR_DEVICE`

I'm not compiling CellularConnectionFSM.cpp as long as the .json does not have "device has CELLULAR".

Files located in features/cellular/framework/targets shall contain #if... #endif to avoid compiling them
They are just missing a define + its related json usage, so that we can get rid of compiling them.

I hope this explanation is more convincing.

Kind regards
Armelle

adustm, 2018-07-25:
Thanks Martin!

0xc0170, 2018-09-04:
> From #7644 it seems that system is able to scan only available components ?
Why the same system couldn't be applied with features ?

cc @ARMmbed/mbed-os-tools  Something to cover in the docs

sg-, 2018-09-17:
Do we have docs written for components yet? The intention is for anything "not in the soc" to be a component which is selectable (add/remove) by the target, libraries or application.

There are gray areas as to what is in the soc and what is available to be wired to the soc and for those cases we should be refactoring from the target to a component for reuse where reasonable or possible.

A target can now be more comprehensive to look like complete board support where an OS api exists.

cmonr, 2018-09-17:
@sg- The main documentation that I know of can be found here: https://github.com/ARMmbed/mbed-os-5-docs/pull/693/files

@AnotherButler could possibly provide something more relevant.

AriParkkila, 2018-09-18:
@cmonr according to that document, for example `feature/cellular` folder should be compiled only when `"target.features": ["CELLULAR"]` is defined?

cmonr, 2018-09-18:
@AriParkkila That sounds correct.

jeromecoutant, 2018-10-02:
@adbridge Why do you close this issue ?

adbridge, 2018-10-02:
@jeromecoutant it's our new automated script reflecting the internal defect status. We are updating our automation to be smarter. Once the update has completed please re-open any issues  that are still relevant (adding a suitable comment) and this will re-open the internal issue also.

jeromecoutant, 2018-10-02:
Please re-open this one...

jeromecoutant, 2018-10-02:
Action is still ARM side

AriParkkila, 2018-10-03:
@sg- do you think cellular and lora should be component of feature?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCELL-1188*

jeromecoutant, 2018-12-13:
Please re-open again this one...

Issue is still present:

````
mbed test -t ARM -m DISCO_L475VG_IOT01A --compile | egrep "QUECTEL|TELIT|AT_"

Compile [  5.5%]: AT_CellularBase.cpp
Compile [  5.6%]: AT_CellularInformation.cpp
Compile [  5.7%]: AT_CellularDevice.cpp
Compile [  6.0%]: AT_CellularNetwork.cpp
Compile [  6.1%]: AT_CellularContext.cpp
Compile [  6.2%]: AT_CellularPower.cpp
Compile [  6.3%]: AT_CellularSIM.cpp
Compile [  6.7%]: AT_CellularStack.cpp
Compile [  7.0%]: AT_CellularSMS.cpp
Compile [  8.4%]: QUECTEL_BC95.cpp
Compile [  8.6%]: QUECTEL_BC95_CellularContext.cpp
Compile [  8.7%]: QUECTEL_BC95_CellularNetwork.cpp
Compile [  8.8%]: QUECTEL_BC95_CellularPower.cpp
Compile [  8.9%]: QUECTEL_BC95_CellularSIM.cpp
Compile [  9.1%]: QUECTEL_BG96.cpp
Compile [  9.2%]: QUECTEL_BC95_CellularStack.cpp
Compile [  9.3%]: QUECTEL_BG96_CellularContext.cpp
Compile [  9.4%]: QUECTEL_BG96_CellularNetwork.cpp
Compile [  9.6%]: QUECTEL_BG96_CellularPower.cpp
Compile [  9.7%]: QUECTEL_BG96_CellularSIM.cpp
Compile [  9.8%]: QUECTEL_UG96.cpp
Compile [  9.9%]: QUECTEL_UG96_CellularContext.cpp
Compile [ 10.1%]: QUECTEL_BG96_CellularStack.cpp
Compile [ 10.2%]: QUECTEL_UG96_CellularNetwork.cpp
Compile [ 10.3%]: QUECTEL_UG96_CellularPower.cpp
Compile [ 10.4%]: TELIT_HE910_CellularNetwork.cpp
Compile [ 10.6%]: TELIT_HE910_CellularContext.cpp
Compile [ 10.7%]: TELIT_HE910.cpp
Compile [ 10.8%]: TELIT_HE910_CellularPower.cpp
Compile [ 10.9%]: UBLOX_AT_CellularNetwork.cpp
[Warning] UBLOX_AT_CellularNetwork.cpp@70,0:  #111-D: statement is unreachable
Compile [ 11.2%]: UBLOX_AT_CellularPower.cpp
Compile [ 11.7%]: UBLOX_AT_CellularContext.cpp
Compile [ 11.9%]: UBLOX_AT_CellularStack.cpp

````

closed 2019-09-26 by ciarmcom

0xc0170, 2019-09-26:
I am not certain why this was reopened as internally it was closed as "rejected", works as it should.

cc @AnttiKauppila

40Grit, 2019-09-28:
Note: The config docs say FEATURES may becompiled in regardless of if they are used.

isaev-d, 2020-10-31:
It should not be in closed status. Sure we can use mbdignore hacks, but it is black magik - like. You can do things, but you wouldn't be happy after all. For example I do not need LoRa, tls, and GSM in my app. I can't just remove whole folders because of some dependencies. Tons of errors while trying to tune! That is weird.
ARM mbed needs config system, any. Just look at NuttX or Zephir.

jeromecoutant, 2020-11-02:
See #13809 for information

0xc0170, 2020-11-02:
@isaev-d The new tools coming to master (CMake based), that one will  fix this.

isaev-d, 2020-11-02:
Oh. thanks.
When I missed CMake?

0xc0170, 2020-11-02:
See https://github.com/ARMmbed/mbed-os/pull/13566