ARMmbed/mbed-os #7458: Avoid compiling features/lora files when not needed

adustm, 2018-07-10:
### Description
Hello,
It looks like LORA feature is not protected with #ifdef LORA_SOMETHING at all.

I am constantly compiling this feature, whatever platform I am using.
For instance DISCO_L475VG
Compile [ 12.9%]: LoRaWANInterface.cpp
Compile [ 13.1%]: LoRaWANStack.cpp
Compile [ 13.2%]: LoRaMac.cpp
Compile [ 13.3%]: LoRaMacChannelPlan.cpp
Compile [ 13.4%]: LoRaMacCrypto.cpp
Compile [ 13.5%]: LoRaMacCommand.cpp
Compile [ 13.6%]: LoRaPHYAS923.cpp
Compile [ 13.8%]: LoRaPHY.cpp
Compile [ 13.9%]: LoRaPHYAU915.cpp
Compile [ 14.0%]: LoRaPHYCN470.cpp
Compile [ 14.1%]: LoRaPHYCN779.cpp
Compile [ 14.2%]: LoRaPHYEU433.cpp
Compile [ 14.3%]: LoRaPHYIN865.cpp
Compile [ 14.5%]: LoRaPHYEU868.cpp
Compile [ 14.6%]: LoRaPHYKR920.cpp
Compile [ 14.7%]: LoRaPHYUS915.cpp
Compile [ 14.8%]: LoRaWANTimer.cpp
Compile [ 14.9%]: LoRaPHYUS915Hybrid.cpp

Could someone make sure the correct #defines are set for this feature, please ?

Kind regards
Armelle

### Issue request type
[ ] Question
[ ] Enhancement  
[X] Bug

0xc0170, 2018-07-10:
Similar to #7457

0xc0170, 2018-07-25:
This is how the build system works. You can use mbedignore file to ignore files, otherwise it builds it all

I'll close this as won't fix (works as designed).

cc @bulislaw

closed 2018-07-25 by 0xc0170