ARMmbed/mbed-os #7467: Testing Custom BlockDevices

AGlass0fMilk, 2018-07-10:
### Description
Hi,
I have written a custom BlockDevice-style driver for an external flash chip my device is using. It would be really nice for mbed to provide a suite of tests that can be easily configured to use a custom block device and ensure it is completely compatible with mbed's API _according to mbed's testing standards._

I would also like to be able to test my block device driver with littlefs so I can make sure there aren't any corner-cases I am missing. I see there are some configurations for the littlefs tests but I haven't found any public documentation about them.

If this exists, please point me in the right direction. If not, I suggest it should be added - it would be a great tool for users to ensure their block device driver is compatible!

### Issue request type
[ x ] Question  
[ ] Enhancement  
[ ] Bug

dannybenor, 2018-07-11:
Please use https://github.com/ARMmbed/mbed-os-example-blockdevice
For littlefs you can use https://github.com/ARMmbed/mbed-os-example-filesystem

dannybenor, 2018-07-11:
Let me know if these helps

AGlass0fMilk, 2018-07-12:
I'll take a look. I was mostly looking to test using your actual unit tests rather than just run an example.

Any idea how to configure the unit tests here: https://github.com/ARMmbed/mbed-os/tree/master/features/filesystem/littlefs/TESTS

@geky?

geky, 2018-07-12:
For block devices, right now we only have one test file, and it is currently copy-and-pasted into each driver with local modifications for the block device:
https://github.com/ARMmbed/spif-driver/blob/master/TESTS/block_device/spif/main.cpp#L34

For the filesystem tests, you can run the littlefs tests on a block device with two defines, MBED_TEST_BLOCKDEVICE and MBED_TEST_BLOCKDEVICE_DECL:
``` bash
mbed test -n features-filesystem-littlefs-tests-* "-DMBED_TEST_BLOCKDEVICE=SPIFBlockDevice" "-DMBED_TEST_BLOCKDEVICE_DECL=SPIFBlockDevice bd(D10, D11, D12, D13)"
```

However, note that these are tests written to test the filesystem. Any errors may be hidden behind filesystem operations, so I would suggest starting with the block device tests.

This is what we have currently. We're currently working on standardizing the tests across the filesystem/block devices, but this has been the first area where we need to test multiple devices independently from the underlying target, so there are several tool changes that have slowed progress.

Let us know if you run into any problems.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTSTOR-485*

closed 2018-10-23 by ciarmcom