ARMmbed/mbed-os #7469: Requires a configuration to enable HSI for STM32L4 family.

Eman869, 2018-07-11:
### Description
Requires a configuration so that user can enable HSI without USE_PLL_HSI. HSI can be the clock source of peripherals, without being the clock source of PLL.

### Issue request type
[ ] Question  
[x ] Enhancement  
[ ] Bug

jeromecoutant, 2018-07-11:
Use case:
- Be able to use HSI for LPUART while PLL is configured with MSI
- Be able to use LSE/LSI for LPUART while RTC is not enabled

bcostm, 2018-07-11:
I would change the title and put `STM32 `instead of STM32L4.

Suggestions:
- add for example ENABLE_HSI and ENABLE_LSE macros in json file 
- configure them in system_clock.c file when the system clock is configured ? or only when needed for example in serial_baud function ?

Eman869, 2018-07-11:
My original idea is to have the HSI, LSE clock enabled when they are needed by peripherals. Or is it too discrete to handle the clock setting?

bcostm, 2018-07-12:
@Eman869 can you please check if the PR #7498 solves this issue ? Thanks.

Eman869, 2018-07-17:
Yes, I had tested PR #7498 and it  is OK to enable HSI and LSE for peripheral LPUART.

bcostm, 2018-07-17:
OK thanks for confirming. Can you please close the issue ?

closed 2018-07-23 by 0xc0170