ARMmbed/mbed-os #7470: mbed Cellular APIs 3GPP PSM

wajahat-ublox, 2018-07-11:
### Description

There is a method `opt_power_save_mode` in `AT_CellularPower` class. It is responsible for controlling the 3GPP Power Saving Mode. I want to know how function call for this method is to be made? To my understanding, the AT_xxxxx classes are not exposed directly to user application and EasyConnect + ConnectionFSM are using methods from AT_xxxxxx classes but currently `AT_CellularPower::opt_power_save_mode` is not used by ConnectionFSM. Can someone explain a little?

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

mirelachirica, 2018-07-12:
I suggest adding it in FSM. 
Either:
`void CellularConnectionFSM::state_init()
{
...
   **_power->opt_power_save_mode(...)** 
    nsapi_error_t err = _power->is_device_ready();
...
`
or in init() after _power is available:
`nsapi_error_t CellularConnectionFSM::init()
{
...
    _power = _cellularDevice->open_power(_serial);
    if (!_power) {
        stop();
        return NSAPI_ERROR_NO_MEMORY;
    }
    **_power->opt_power_save_mode(...)** 
`

wajahat-ublox, 2018-07-12:
Thanks for the answer but it is unclear to me as how this allows the user to control the power saving mode? Looks like it will always be enabled if i call the method from `CellularConnectionFSM::state_init()` or `CellularConnectionFSM::init()`

mirelachirica, 2018-07-12:
I see. It would be needed to be used independently in the application at any time, right?

wajahat-ublox, 2018-07-12:
I suppose so because not all user applications would want to enable the power saving mode.

philwareublox, 2018-07-12:
Some applications might want to change the power saving mode from time to time.

mirelachirica, 2018-07-12:
EasyCellularConnection doesn't provide direct access to opt_power_save_mode(), at the moment. It needs to be enhanced to support this alone as quick fix or support a mechanism to give direct access to power interface to the application. Would you need a proposal patch for the first solution, the quick fix?

wajahat-ublox, 2018-07-13:
What would be the difference between first and second solution as both will allow user application to control the PSM, is that correct?

mirelachirica, 2018-07-13:
Yes, no difference from user's perspective.

wajahat-ublox, 2018-07-19:
Since this [commit](https://github.com/u-blox/mbed-os/commit/d9a99b0a455924e81ab329624ca44d9cad88851f), the _power object is deleted as soon as the device is ready so i am not sure how either of the solutions can work.

mirelachirica, 2018-07-19:
You are right, that is the next problem. I will have to discuss this issue with the team after the holiday ~2 weeks, plan and come up with a full solution to this problem.

wajahat-ublox, 2018-07-19:
Thank you. Since your team will be working on it, I would like to draw your attention towards a related problem. If you think it should be discussed in a separate thread then please let me know.

Along with 3GPP PSM, some modems support their own proprietary power saving mode as well. In order to avoid confusion between 3GPP PSM and proprietary PSM, i will be calling the proprietary PSM as "idle mode". I want to get your opinion as how vendors can implement driver for such an idle mode and integrate with the Cellular APIs?

In terms of implementation, the vendor will provide a method to enable or disable the idle mode and another method to do the "wake-up". Now in order to wake up the modem, some modems would require a pin toggle while other would require an activity on modem uart. In the latter case, this would mean that before sending any AT command, modem needs to be woken up by appending an extra character before the actual command. This might require changes at lower layers of the Cellular APIs architecture.

wajahat-ublox, 2018-08-13:
A friendly reminder.

mirelachirica, 2018-08-15:
One approach is getting access to cellular device and serial from EasyCellularConnection: https://github.com/ARMmbed/mbed-os/pull/7795 and through device, access to power/open_power().
```
EasyCellularConnection *ecc = reinterpret_cast<EasyCellularConnection*>(iface);
CellularPower *power = ecc->get_device()->open_power(ecc->get_serial());
power->opt_power_save_mode(0, 0);
```
What do you think?

wajahat-ublox, 2018-08-27:
Looks like it does solve the problem. Thank you