ARMmbed/mbed-os #7476: Why does the semihosting API have unused arguments in `semihost_read()` and `semihost_write()`?

danieljabailey, 2018-07-11:
### Description
Why does the semihosting API have unused arguments in `semihost_read()` and `semihost_write()`?

In `platform/mbed_semihost_api.[ch]` are functions with these signatures:

```int semihost_read  (FILEHANDLE fh, unsigned char* buffer, unsigned int length, int mode);```
```int semihost_write (FILEHANDLE fh, const unsigned char* buffer, unsigned int length, int mode);```

As far as I can tell these functions don't make use of the `int mode` argument.
I haven't checked particularly thoroughly but I can't find any docs for this argument.

Is it just here for historical reasons?

### Issue request type

[x] Question  
[ ] Enhancement  
[ ] Bug

0xc0170, 2018-07-12:
mode for read/write mode - I can't find a reference for it neither. I understand the mode argument in the open but not for read/write. We will check.

SenRamakri, 2018-07-18:
@danieljabailey and @0xc0170 - I did some research around this and it appears its carried over from older mbed-os versions. So, its more like historical reasons I would say and may be an attempt to keep backward compatibility. We may clean this up in future after more investigation but at this point I would close this issue as we have answered the question.

closed 2018-07-18 by SenRamakri