ARMmbed/mbed-os #7484: greentea client start without waiting for sync

RobMeades, 2018-07-11:
### Description

As mentioned in #7441, I need to be able to run my mbed tests without `stdin` (just `stdout` going via SWO as I have no spare serial port to dedicate to testing).

Aside from hacking the `greentea` client, is there a way to persuade `greentea` to begin running tests immediately, without waiting for the sync string (which I can't possibly send)?  I know that you can ask `mbed test` to start without _sending_ the sync message, so I'm hopeful...

### Issue request type

[x] Question  
[ ] Enhancement  
[ ] Bug

aashishc1988, 2018-08-14:
@RobMeades I am cc'ng couple of people here, they can correct me if I am wrong about my understanding of the architecture. 
So, right now, there is not a way for greentea client to start without waiting for sync. We have a blocking call in greentea setup that needs a sync packet. Considering there is no stdin, one way as suggested by @studavekar is that we can ifdef that part of the code where it is waiting for the sync by implementing some sort of build compilation flag, but again that is something you can experiment on your local setup. I am not sure supporting it officially (adding a mechanism in our tools/build setup) is something we want to do, as there are repercussions if we dont wait for the host to sync (missing uart messages is one of them), and adding this sort of our support might also tempt people to use this feature to save startup time for a test, which might lead to undesirable test results. 

cc: @cmonr @studavekar @theotherjimmy

cmonr, 2018-08-18:
@aashishc1988 @RobMeades I don't this would be high on our priority list if we decide to attempt it.

The sync mechanism is in place to confirm bidirectional communication between the host and device under test.

@RobMeades If you don't have stdin available in your system, you're looking to run output-only tests? This feels like quite a departure from how Greentea and HTRun were designed to run. Is this correct?

aashishc1988, 2018-10-02:
@RobMeades , thoughts?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-255*

aashishc1988, 2018-10-17:
@RobMeades I am going to close this issue as the current request can be fulfilled by making changes locally on your setup, and the request is quite different from how htrun and greentea were designed to run. Please feel free to reopen the issue if you disagree and we can continue further discussion.

closed 2018-10-17 by studavekar

aashishc1988, 2018-11-13:
@RobMeades you might be interested in this, https://github.com/ARMmbed/greentea/issues/103