ARMmbed/mbed-os #7488: NVSTORE_NUM_AREAS is not configurable parameter

teetak01, 2018-07-12:
Currently NVSTORE_NUM_AREAS has been hardcoded to be two:

https://github.com/ARMmbed/mbed-os/blob/master/features/nvstore/source/nvstore.h#L68-L69

We would need this parameter to be configurable. It would make sense to allow both 1 and 2 as valid values. For some products like Mbed Cloud Client it makes sense to always have two sectors, but for some other products like Mbed Client Lite, the second area is waste of flash.

[ ] Question  
[X] Enhancement  
[ ] Bug

teetak01, 2018-07-12:
@yogpan01 @JanneKiiskila @davidsaada

davidsaada, 2018-07-12:
@teetak01
Reducing the number of NVStore areas to one is not possible because of the following reason: The way NVStore works, when setting a new value for a key, it writes it on the next free space of the active area, overriding the older values previously written. When the area gets filled, a GC (garbage collection) process is invoked, which is the process of copying only the newer values to the second area, and turning it to the active one. Having just one NVStore area means we can't perform GC (as we can't hold all active key values in memory).
Now, even if we assume that the area will never get filled (which is a dangerous assumption), we can't do without GC. This is due to the fact NVStore is also immune to power failures: If we have a power failure during a write, the information at the end of the sector will be corrupt. Right now, the following init process can salvage all information written in flash prior to this write using the GC process. If we have one area only, then the only way to recover from such a scenario is by erasing the single area and losing all our information, which is unacceptable.

davidsaada, 2018-07-12:
> @davidsaada we had the understanding that if we are using NVStore to store keys only once and perform only read operation afterwards we can get away with reserving 2 sectors for it. But is GC case also applicable for read-only operation scenario? NVStore can have default 2 regions reserved but this can be made configurable so that applications which have read-only use case can decide if 1 region is enough for them and they can save on precious flash memory for application usage.

@yogpan01 In that case, what you are suggesting is that one NVStore area -> NVStore read only mode. Correct?

bulislaw, 2018-07-12:
> The solution, as discussed with the the tools team, was to relocate the managed boot loader after the first sectors, which are typically smaller than the last ones, and can populate NVStore. I don't know where this stands. @sg- @bulislaw - Can you shed some light?

@SenRamakri @theotherjimmy are you guys aware of this discussion?

teetak01, 2018-07-12:
Generally Cloud Client / Client Lite allocate the smallest sectors for NVSTORE/SOTP, but customers might want to use other regions also. For some boards the smallest available sector size is 16 kB.

https://github.com/ARMmbed/mbed-cloud-client-example/blob/master/mbed_lib.json#L75-L80

dannybenor, 2018-07-12:
@teetak01 A read-only NVStore is a valid requirement. Please ask add it to the product requirements

theotherjimmy, 2018-07-12:
@davidsaada While we did discuss that as an option, I was not left with the impression that it was a requirement.

teetak01, 2018-08-21:
cc: @JanneKiiskila we need to check if this needs IOTPREQ.

closed 2018-08-21 by teetak01