ARMmbed/mbed-os #7492: Priority of mbed-os tick undefined

david-reimondez-tridonic-com, 2018-07-12:
### Description
If we look at the following function, in file: mbed-os/targets/TARGET_STM/hal_tick_32b.c 

HAL_StatusTypeDef HAL_InitTick(uint32_t TickPriority)

It takes the priority as parameter but this parameter is never used. The interrupt is activated but no priority is set. Therefore the default priority is used, which is normally the highest one.

Is this behavior the expected one? With which priority level is mbed-os supposed to work?
Is it desired to have the tick interrupt with the highest priority level?

Thanks a lot for your help!

------

    [X] Question
    [ ] Enhancement 
    [ X] Bug 

---

ARM: @MarceloSalazar @karsev 
ST: @bcostm 

FYI: @markus-becker-tridonic-com @daniel-cesarini-tridonic-com @javier-moreno-tridonic-com

MarceloSalazar, 2018-07-12:
@bulislaw can you look into this asap please

bulislaw, 2018-07-12:
@ARMmbed/team-st-mcd

bcostm, 2018-07-12:
> It takes the priority as parameter but this parameter is never used.

This is normal. This parameter is used in the default function definition for the Systick configuration. As this function is declared "weak" we cannot remove its prototype.

> Therefore the default priority is used, which is normally the highest one. Is this behavior the expected one? 

 I suppose yes as this interrupt is used for timeouts, schedule events etc... I will add also that there is no possibility to pass a priority parameter to the us_ticker_init function.

david-reimondez-tridonic-com, 2018-07-13:
We have another interrupt which must be executed every 52us and the systick interrupt is causing a jitter.
Is it possible to decrease the priority of the the Systick interrupt?

bcostm, 2018-07-13:
For the Systick interrupt, I found in https://github.com/ARMmbed/mbed-os/blob/master/rtos/TARGET_CORTEX/rtx5/Source/os_systick.c#L34 that the priority is set to the lowest value (255). But I am not sure, as the `SYSTICK_IRQ_PRIORITY` macro is maybe set somewhere else ?

For the us_timer interrupt we didn't have configured the priority (excepted for the STM32F0 here https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/us_ticker.c#L101 ). Is it this interrupt priority that you would like to be lower ?

david-reimondez-tridonic-com, 2018-09-03:
Well, we would like to understand what is the correct priority and what are the side effects if we change it. What happens if we preempt this interrupt every 52 us with another interrupt which takes around 3 us? Is this fatal for the system?
In the current version we are using we do not have the file "us_ticker.c" instead we have "us_ticker_32b.c"

bcostm, 2018-09-03:
> In the current version we are using we do not have the file "us_ticker.c" instead we have "us_ticker_32b.c"

We have refactor the ticker files recently and the us_ticker_32.c/16.c files have been merged in the us_ticker_16b.c file. This is why you can't see this file now.

bulislaw, 2018-09-26:
If you have issues with SysTick introducing jitter to your other interrupt, it's most likely it will happen the other way if you change the priorities (your interrupt delaying SysTick). 3us is relatively short time, I'm trying to think if in this case it can accumulate and introduce bigger inaccuracies in the system over  time and I don't think so. The scheduler doesn't guarantee this kind of accuracy anyway. 

Some more details: http://arm-software.github.io/CMSIS_5/RTOS2/html/theory_of_operation.html#kernelTimer

@c1728p9 any thoughts?

c1728p9, 2018-09-26:
> Is it desired to have the tick interrupt with the highest priority level?

Yes, most interrupts should be left as high priority by default. The more priority levels you use the more interrupts can stack on top of each other. With multiple priority levels the worst case stack usage is the sum of the worst case of each priority level. This can be problematic with the typical interrupt stack of 1KB.

> Well, we would like to understand what is the correct priority and what are the side effects if we change it. What happens if we preempt this interrupt every 52 us with another interrupt which takes around 3 us? Is this fatal for the system?

The correct priority for mbed is typically the highest. This prevents high stack usage as mentioned above. Regardless of priority interrupt safe mbed code is typically protected by a critical section. This means that mbed interrupt handlers are safe to use at any priority, but changing the priority likely will not prevent jitter, as the critical section protecting it will prevent all interrupts and take nearly as long.

The one place it is beneficial to use low priority levels is when the code is specifically meant to be pre-empted. Examples of this are the handlers PendSV and SysTick, which are set to a low priority by default. Neither of these disable interrupts so you will see a decrease in jitter with lower priority. The mbed us ticker code on the other hand does all its computations in a critical section, so there is not a big advantage to lowering its priority.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-5809*

david-reimondez-tridonic-com, 2018-10-10:
Thanks for the feedback. It makes sense but I am little bit confused by the last paragraph of the last comment:

"The one place it is beneficial to use low priority levels is when the code is specifically meant to be pre-empted. Examples of this are the handlers PendSV and SysTick, which are set to a low priority by default"

But we observed the opposite, that the SysTick interrupt priority is not set, and it gets by default the highest. Therefore my question what happens if we reduce the priority? Should we set to the lowest?

Did I missunderstood the last comment?

Thanks again!

MarceloSalazar, 2018-10-26:
@c1728p9 @bulislaw can you please have a look? Thanks!

c1728p9, 2018-10-26:
Hi @david-reimondez-tridonic-com SysTick should have the lowest priority since this is what the [OS is expecting](https://github.com/ARMmbed/mbed-os/blob/master/rtos/TARGET_CORTEX/rtx5/Source/os_systick.c#L34). This is set as part of the OS initialization sequence started by `osKernelStart()`.

To confirm that the interrupt priority was correct on mbed-os I attached a debugger to a NUCLEO_F401RE and found that SysTick is correctly set to the priority 0xFF by the OS and stays at that value. I did this at the tip of master as of now - 3f5eeac29d4292a4f0a7c95b537437bce32a56fe.

> But we observed the opposite, that the SysTick interrupt priority is not set, and it gets by default the highest. 

@david-reimondez-tridonic-com are you seeing some scenario where this value is incorrect? If so could you provide me with details on how to reproduce it?

david-reimondez-tridonic-com, 2018-11-20:
According to the "STM32F3 and STM32F4 Series Cortex®-M4 programming manual" (PM0214 - section 4.3.7) the "Interrupt Priority Registers" (NVIC_IPRx) have as reset value 0x0. Therefore the default priority is the maximum (and also what I have experienced while debugging). 
If as you mentioned, the OS change the value of these registers, where is this done? I could not find this...

kjbracey-arm, 2018-11-20:
Priority is set either [here](https://github.com/ARMmbed/mbed-os/blob/9aef9d36612ea4f75f93d254ee442ded0a1ffde0/rtos/TARGET_CORTEX/rtx5/Source/os_systick.c#L55), or [here](https://github.com/ARMmbed/mbed-os/blob/9aef9d36612ea4f75f93d254ee442ded0a1ffde0/rtos/TARGET_CORTEX/SysTimer.cpp#L74) when using a systick substitute.

There is an added wrinkle when using MBED_TICKLESS in that the low-power timer is used instead, and that interrupt will be maximum priority, but then triggering a minimum priority one, [here](https://github.com/ARMmbed/mbed-os/blob/master/rtos/TARGET_CORTEX/SysTimer.cpp#L183).

Interrupt latency via lp ticker was tripping us up [in this issue](https://github.com/ARMmbed/mbed-os/issues/8714). I didn't get an answer on that issue to my question of why MBED_TICKLESS doesn't just use SysTick while running.

david-reimondez-tridonic-com, 2018-11-20:
I did not refer to the Systick interrupt but to the time basis created from a hardware timer in the file:
mbed-os/targets/TARGET_STM/hal_tick_32b.c

Where is this priority set? If it is not set in the software, then at least in STM32 we will get the maximum. Therefore my question if this was expected..

Sorry for the missunderstanding and thanks for your support!

kjbracey-arm, 2018-11-20:
Okay, that is part of providing the `us_ticker_api.h` HAL functions for STM, but I don't see `HAL_InitTick` being called anywhere - looks like dead code?

Mbed OS doesn't itself touch the priority of any interrupts apart from the 3 associated with the RTOS - SVPendSV SVCall and SysTick, as far as I know, so we would normally expect that us_ticker interrupt to remain at the hardware default of maximum, which is what you're seeing.

The target HAL code is free to adjust the priorities of interrupts other than the RTOS-related ones, which must remain minimum priority. But as Russ said, using multiple levels might cause some unpleasant side-effects, including increased IRQ stack usage.

Personally I think it's a bit unfortunate that the hardware defaults to "maximum", which means if we just don't touch it we end up with everything at maximum like now, and it's quite a pain to try to raise the priority of an interrupt. Would be nice if we started off with everything at "medium" by default.

For your example, I think this would be fine - if you have a really high-priority interrupt, you could whiz through the table, set all other interrupts with priority 0x00 down a notch to 0x10, and have your high-priority one at 0x00.

But the catch is that there is a lot of code using `enter_critical` which masks off all interrupts - most such blocks should be short, but there might be long-running ones (like #8714). I would not realistically expect you to 100% reliably get better than 100us latency on an IRQ as a result.

"every 52 us with another interrupt which takes around 3 us" would be quite reasonable load for an IRQ handler, but you will have latency spikes. Your period may not be a terribly smooth 52us, although it would depend on what other code and drivers were doing.

If this is supposed to be some sort of real-time pseudo-DMA, that needs <10us latency, then you ideally should use NMI rather than IRQ - that will give you great latency, but you won't be able to call any OS code inside the NMI handler - you'd penetrate all the usual protection locks. You'd have to signal any completion or status updates by generating a software "downgrade" IRQ. (Or maybe just rely on foreground code polling volatile status variables that you're updating.)

(ARM-A has better support for this with its FIQ mode, and even 6 dedicated banked CPU registers for the FIQ handler to push latency down even further. ARM-M is somewhat simpler, but NMI is its equivalent).

Another fiddly alternative which I know a couple of targets use is to replace "enter_critical" with something that only masks off interrupts up to a certain level. That effectively makes higher-prioirity interrupts "software non-maskable", achieving much the same effect as the hardware NMI.

jeromecoutant, 2018-11-20:
> but I don't see HAL_InitTick being called anywhere - looks like dead code?

HAL_InitTick is called in HAL_Init function, called in mbed_sdk_init

kjbracey-arm, 2018-11-20:
Ah, okay, was looking in the higher levels. So individual STM HALs have a default weak HAL_InitTick that configures SysTick with normally minimum priority, but that's been replaced globally with something that instead configures the timer used for `us_ticker_api`. So SysTick is left alone - the HAL ticker functions aren't used - and it doesn't use the priority given to HAL_InitTick, as that wouldn't be what we expect for the us_ticker IRQ.

david-reimondez-tridonic-com, 2018-11-20:
Thanks for the answers

As conclusion, do you consider this an error in the HAL implementation, the fact that the interrupt is not defined?

Regarding the proposals offered, could you provide some example code on how to implement both solutions:
- Using NMI
- Overloading enter_critical

Thanks a lot

kjbracey-arm, 2018-11-20:
Gah, browser crash lost that answer.

> do you consider this an error in the HAL implementation, the fact that the interrupt is not defined?

Which interrupt?

No, don't think there's anything wrong with the HAL. It's not touching SysTick at all, and it's leaving the others including the `us_ticker_api` one at maximum, which is expected.

Don't have any examples to hand - you're in quite specialised territory.

For NMI it appears the NVIC can't route arbitrary external interrupts to NMI? (Unlike ARM-A GIC, which can choose IRQ or FIQ for each, IIRC). So availability is going to be chip-specific. And it looks like that's not available on STM - the manual I found has them hardwiring NMI to be some sort of "emergency failure" interrupt. Using the maximum priority to break into any software, rather than for speed.

For overriding enter_critical, search the tree for `hal_critical_section_enter`. You'll see nRF52 either passing it to an underlying OS or masking certain NVIC interrupts. What I suggested above would be an alternative that sets `BASEPRI` to do it by priority level. Any of those options will require `hal_sleep` to be modified too - for initial testing you can make it a null function. If you ever go down this path I could explain how to make it work properly later.

kjbracey-arm, 2018-11-20:
But all this is fiddly - don't attempt until you find that the normal IRQ setup isn't working with just a priority tweak. It may be okay on your system with the devices you're using and software you're running.

kjbracey-arm, 2018-12-18:
I don't think there are any outstanding questions. Okay to close?

cmonr, 2018-12-19:
@ARMmbed/team-tridonic @ARMmbed/team-st-mcd  ?

MarceloSalazar, 2019-01-15:
Ticket considered solved. Reopen if required.

closed 2019-01-15 by MarceloSalazar