ARMmbed/mbed-os #7496: Add new function to read Active-Time and Periodic TAU

philwareublox, 2018-07-12:
There is a method opt_power_save_mode in AT_CellularPower class which allows the application to use the 3GPP "Power Save Mode" Feature. However, the values provided are only a request to the network. The network may not actually allow those values to be used. 

The actual values which are provide back to the UE are contained in the CEREG response, when it's mode is '4' or '5'.
+CEREG: <stat>[,[<tac>],[<ci>],[<AcT>][,<cause_type>,<reject_cause>[,[<Active-Time>],[<Periodic-TAU>]]]]

At the moment there are no functions in the CellularAPI which allow you to read these values. 

[ ] Question  
[X ] Enhancement  
[ ] Bug

AriParkkila, 2018-07-26:
@philwareublox can you elaborate please, how an application could actually use PSM time?

AriParkkila, 2018-07-26:
@philwareublox if PSM time what the application requested was not approved by the network then what the application should do otherwise?

I guess that in typical case cloud service should try to contact the application based on the application's PSM values. Do you have in mind some kind of case when cloud service wants to wake up the application?

There is also a problem that PSM values may change on each RAU/TAU (and probably also on cell reselection) so how frequently the application should report the values to cloud service?

philwareublox, 2018-07-26:
If the device is in PSM then the cloud service cannot wake up the module. 

The Timers can change at any time, so I would say the application should update their cloud service when it changes or on the very first registration. 

I doubt the timers will change regularly though, as these are very important to the network and device. For low power devices you can't have the network just changing timers whenever it wants on the next TAU update.

If the application's requested timers are not allowed, or they are changed then the application should send a message to their cloud service to inform the company to question the Network Operator. 

There is nothing the application can do to 'fix' the timers as the modem must follow the protocol.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCELL-1176*

closed 2018-11-30 by ciarmcom