ARMmbed/mbed-os #7499: Regression seen in Blinky application on Target nRF52840_DK

naveenkaje, 2018-07-12:
### Description
The blinky application did not work with nRF52840_DK with ARM compiler on master branch’s tip. See details of the commits and tools used below.

- Target: NRF62840_DK
- Tool chain: ARM Compiler 5.06
- Tools used: mbed cli, built with command "mbed compile -m NRF52840_DK -t ARM”
- SHA: f4382d95b886482a707cf1249a16fd6d260f16b6  (tip as of 7/11)
- Steps to reproduce: compile blinky using the above commands and flash on nRF52840 dev kit. The blinky does not work as expected. Use GCC_ARM compiler instead, and it works.

Other useful information:
We found out that it worked with mbed-os-5.9.0 (#866850acc15e86cd4ac11bf4404078a49f921ddd), it worked as expecte on nRF52840

The tip (SHA: f4382d95b886482a707cf1249a16fd6d260f16b6  works for blinky with nRF52_DK)

Please feel free to reach out to me if you need further information.

### Issue request type
[ ] Question  
[ ] Enhancement  
[ x] Bug

naveenkaje, 2018-07-12:
I forgot to mention that the application builds and works fine on GCC_ARM tool chain.

marcuschangarm, 2018-07-12:
This should fix it: https://github.com/ARMmbed/mbed-os/pull/7502

adbridge, 2018-10-04:
Fixed by the aforementioned PR

closed 2018-10-04 by adbridge