ARMmbed/mbed-os #7506: Using Nordic NRF52 Softdevice S132 v3.1.0 in mbed-os builds prior to mbed os 5.9.

yogeshk19, 2018-07-13:
### Issue request type
[ X] Question  
[ ] Enhancement  
[ ] Bug  

### Description
    - NRF52_DK (Nordic NRF52832)
    - GCC ARM?
    - MBED-CLI ?
    - MBED-OS 5.5.4?

Building BLE applications for the Nordic NRF52 (NRF52832) platform in MBED OS 5, prior to MBED OS 5.9 was built against Softdevice SDK 11 and the Softdevice S132 v2.0.0. My question is how do we upgrade this to build against a later softdevice version say v3.1.0. As the newer BLE modules such as Rigado's BMD-300 and other BLE modules come pre-loaded with the later versions of the soft device?

What steps do I need to take to generate an BLE central application targeted for the NRF52832 platform built using mbed-cli , which uses soft device S132 v3.1.0 and mbed os 5.5.4?

Thanks,
Yogesh

yogeshk19, 2018-07-16:
I was wondering anyone from the MBED BLE team had any thoughts on this, as we are completely blocked and our production release of our product is delayed because the Nordic based BLE module was upgraded to use S132 v3.1.0 of the softdevice and the mbed builds are using S132 v2.0.0 softdevice until mbed os 5.9 where the version of the softdevice being used jumped to S132 v5.0.0.

If there is a way to create a priority ticket to handle these kind of questions please let me know and I will create one.

Thanks,
Yogesh

0xc0170, 2018-07-17:
@ARMmbed/mbed-os-pan 

> What steps do I need to take to generate an BLE central application targeted for the NRF52832 platform built using mbed-cli , which uses soft device S132 v3.1.0 and mbed os 5.5.4?

@marcuschangarm Can you review?

TacoGrandeTX, 2018-07-17:
Hi Yogesh,

You are correct - we support the Nordic SoftDevice v2.0.0 (from the Nordic 11.0 SDK) and we just added support for SoftDevice v5.0.0 from the Nordic 14.2 SDK.  We have no plans to provide support for the older 3.x or 4.x SoftDevices.  However we are interested in helping you if we can.  Please email us from your employer's email domain at support@mbed.org and we'll continue the dialogue.

Thank You,
Ralph

0xc0170, 2018-07-18:
Thanks @TacoGrandeTX , this should be now resolved

closed 2018-07-18 by 0xc0170