ARMmbed/mbed-os #7511: Invalid characters after UART reinitialization

aashishc1988, 2018-07-13:
**Target**:
NUCLEO_F103RB
**Toochain**:
Arm
**MbedOS**:
mbed-os-5.9.0-rc3
**Test case**: 
mbedmicro-mbed\call_before_main

While running one of the mbed tests on the NUCLEO_F103RB, we noticed an issue where htrun complained about receiving a byte it cannot decode. The test ran fine on other targets. Narrowing it down further, it seemed like the problem is on the host target. Whenever we flush out the stdout using '\n' or fflush, and the uart gets reinitialized for next stream of data, the very first character it sends is something unusual ('255', or '254') which is causing the issue.

I am attaching snapshots of the data captured from logic analyzer, and you can see the decay on analog during reinitialization.

[image]

In this snapshot, you can see the ascii value of the character sent right after reinitialization ('255')
[image]

This behavior is reproducible each time. 

Steps to reproduce
mbed test -m NUCLEO_F103RB -t ARM -n tests-mbedmicro-mbed-call_before_main -v

[ ] Question  
[ ] Enhancement  
[X ] Bug  

@ARMmbed/team-st-mcd 
@screamerbg

bcostm, 2018-07-16:
I looked in our internal CI test history and didn't see any problem with this test from mbed-os-5.6.6 to mbed-os-5.9.1.

I tested the NUCLEO_F103RB on `master` and I confirm that the test fails now:

```
Exception in thread Thread-3:
Traceback (most recent call last):
  File "c:\python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "c:\python27\lib\threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "c:\python27\lib\site-packages\mbed_greentea\mbed_greentea_cli.py", line 489, in run_test_thread
    verbose=verbose)
  File "c:\python27\lib\site-packages\mbed_greentea\mbed_test_api.py", line 318, in run_host_test
    returncode, htrun_output = run_htrun(cmd, verbose)
  File "c:\python27\lib\site-packages\mbed_greentea\mbed_test_api.py", line 134, in run_htrun
    htrun_output += line.decode('utf-8')
  File "c:\python27\lib\encodings\utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xfe in position 63: invalid start byte
```

jeromecoutant, 2018-07-16:
Hi

I also tried this test, and result is depending on greentea version...

The exception that @bcostm has, comes with greetea version 1.4.0

With greentea version 1.3.3 test is OK even with the issue during greentea synchro:

[1531725281.70][CONN][RXD] MBED: mbed_main() call before main(▒mbedmbedmbedmbedmbedmbedmbedmbed

With #7512 , test is OK with all greentea versions

bcostm, 2018-07-16:
Seems related to PR https://github.com/ARMmbed/greentea/pull/271

0xc0170, 2018-07-17:
@cmonr Please review

cmonr, 2018-07-17:
@aashishc1988 @jeromecoutant @bcostm Could y'all post the output of the following commands?
```
python --version
pip list --verbose
```

I think htrun recently underwent a release, so it's possible the versions that y'all have installed are out of date since the Py3 fixes are really recent.

jeromecoutant, 2018-07-17:
I am still using python 2.7

mbed-cli                      1.7.5       c:\python27\lib\site-packages pip
mbed-greentea                 1.4.0       c:\python27\lib\site-packages
mbed-host-tests               1.4.0       c:\python27\lib\site-packages
mbed-ls                       1.4.6       c:\python27\lib\site-packages

aashishc1988, 2018-07-17:
Here is the output
Python: 2.7.11
mbed-cli          1.7.3
mbed-host-tests   1.3.1
mbed-greentea     1.4.0
mbed-ls           1.4.6

I have tried updating htrun to latest release but still saw the failures.

aashishc1988, 2018-07-19:
@jeromecoutant 
I was stepping through the code to see where the voltage drop happens after we send \n, and I narrowed it down to
File: mbed-os/targets/TARGET_STM/serial_api.c
> #if defined (USART2_BASE)
>     if (obj_s->uart == UART_2) {
>         __HAL_RCC_USART2_FORCE_RESET();
>         __HAL_RCC_USART2_RELEASE_RESET();
>         __HAL_RCC_USART2_CLK_ENABLE();
>     }
The call to function __HAL_RCC_USART2_FORCE_RESET();, brings the voltage down to 0, and then the call to init_uart(obj) later pulls it up. We were wondering if there is any software enable pull-up on tx/rx ports that we have available, because we manually hooked up a pull-up resistor to rx, and we don't see the issue anymore (observed via logic analyzer). Let me know if you have any more questions for me regarding this.

jeromecoutant, 2018-07-20:
Hi

Does it work if you remove all FORCE_RESET in serial_init function ?

@LMESTM

aashishc1988, 2018-07-20:
@jeromecoutant 
Yes, it works if I remove FORCE_RESET

bcostm, 2018-07-27:
I confirm that it works also by adding a hardware pull-up on the RX pin.

In the NUCLEI_F103RB PeripheralPins.c file, we configure the pull-ups on the SERIAL pins and also inside the serial_init function. So, I don't understand why it doesn't work ?

I made a trial by moving the pull-ups configuration before the force_reset in the serial_init function, but the test is still fail....

I have a question: would it be still require that we remove the FORCE_RESET after the correction done in htrun PR 207 ?

aashishc1988, 2018-07-27:
@bcostm PR 207 will handle non-unicode characters gracefully now, thats about it. The test wouldnt crash and would print out a message that states it encountered a non-unicode character, but we would ideally like this issue fixed from partners side if we can, unless for some reason the current UART behavior is considered desired/normal.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-263*

closed 2019-02-05 by ciarmcom