ARMmbed/mbed-os #7513: UBLOX-NINA-B1 (nRF52832) Serial API putc not working

edablan, 2018-07-13:
### Description

    - MTB_UBLOX_NINA_B1 (nRF52832)
    - IAR (8.20.2) using Segger J-LINK debugger
    - mbed-cli version (1.3.0)
    - mbed-os 62f8b922b

Using the provided mbed Serial API example:
`#include "mbed.h"
 
Serial pc(USBTX, USBRX); // tx, rx
 
int main() {
    pc.printf("Hello World!\n\r");
    while(1) {
        pc.putc(pc.getc() + 1); // echo input back to terminal
    }
}`

The getc() works fine and receives the sent character. However, putc() returns an -1 error when it tries to transmit. Replacing SERIAL API putc() and getc() with stdio.h putchar() and getchar() works correctly to echo characters:

`Serial pc(USBTX, USBRX); // tx, rx
 
int main() 
{
    pc.printf("Hello World!\n\r");
    while(1)
    {
        putchar(getchar()); // echo input back to terminal
    }
}`

So I know my UART workes fine since the above replacement works, its just the Serial API calls to putc() that all return errors. I also tried using Serial API write() and it works in the below example. Something is specifically wrong with Serial API putc()

`Serial pc(USBTX, USBRX); // tx, rx
 
int main() 
{
    uint8_t string[] = "Help Me/n/r";
    
    pc.printf("Hello World!\n\r");
    while(1)
    {
        pc.write(string, strlen((char*)string), NULL);
    }
}`

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

edablan, 2018-07-16:
I also verified this bug using a EVK-NINA-B1, not just my prototype board

edablan, 2018-07-17:
Ok i built this using IAR 7.80.4 and the problem goes away. Since mbed is only offically supporting 7.8 I will go ahead and close this bug.

closed 2018-07-17 by edablan

0xc0170, 2018-07-18:
Thanks for reporting back. We will check this once we update to IAR 8