ARMmbed/mbed-os #7514: Singletest automated testcase MBED_24 "Timeout Int us" is failing

shrichippabalaji, 2018-07-14:
### Description
- Testcase should take 1sec to complete, but taking 1.23sec.
- Implemented Using 16bit timer with 3Mhz timer and 24Mhz clock with Cortex M0 Custom board.
- The similar code on other target (Cortex M4, clock 60Mhz) by same vendor and TRM works fine.
- Any inputs to debug will be helpful, which part of code of us_ticker should i look on.
- Except MBED_24 all other singletest and greentea  tests are passing.

Carried Out Experiments:
- Changed 1sec calculations in testcase from 500us*1000 to 5000us*100- OK. (1000*500-OK-1.1sec, 2000*250-OK-1.040sec, 5000us*100-OK-1.03s)
- 500us is producing pulse of 560us with timeout object but exactly 500us with ticker object.
- Used individual timers for SetInterrupt and free-running timer - Same Result (FAIL)
- Checked the code for timer without mbed - OK
- Wait() for 500us is producing delay of around 505us to 570us, but it is polling, and ticker_read function itself is taking about 60us to complete.
- Testcase is tested on mbed-os with and without timeout object 
    (copy of ticker APIs, calling set_interrupt directly with value-[500*3 + curr_tick]) with same ticker code.
    with timeout is giving 1.220sec and without timeout object is giving proper 1.0 sec.
### Issue request type
[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-07-17:
Can you clarify:

- how do you run the test (is it using singletest.py)? The name MBED_24 indicates that
- steps to reproduce the failure locally

jamesbeyond, 2018-07-24:
Ping @shrichippabalaji,
could you clarify your issues by answering those questions: 
> * how do you run the test (is it using singletest.py)? The name MBED_24 indicates that
> * steps to reproduce the failure locally

closed 2018-07-26 by shrichippabalaji