ARMmbed/mbed-os #7521: STM32 RTC initialization failed

pilotak, 2018-07-16:
### Description
When I run following code and disconnect power supply and put it back on after a second, most of the time i get an error, or the app doesn't start at all, nothing helps to recover but pressing reset button - disconnecting and connecting power supply again would not restart it, it remains in "freeze" mode.

```
++ MbedOS Error Info ++
Error Status: 0x80FF0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x8000CC1
Error Value: 0x0
Current Thread: Id: 0x2000172C Entry: 0x8001741 StackSize: 0x1000 StackMem: 0x20001778 SP: 0x20002668
-- MbedOS Error Info --
RTC initialization failed
```

Here is the code:
```cpp
#include "mbed.h"

time_t start_timestamp = 0;

void getTime() {
    time_t timestamp = time(NULL);
    printf("uptime: %lu\n", timestamp - start_timestamp);
}

int main() {
    printf("set time\n");
    set_time(1514764800);  // 1.1.2018
    time_t timestamp = time(NULL);
    start_timestamp = timestamp;

    while (1) {
        printf("alive\n");
        Thread::wait(1000);
        getTime();
    }
}
```

running latest OS: be215a310d51c694a992e913f6757e96c397cefd
Target: STM32F412ZG

When i run the same code over 19c6f3b316dd90c72c4f7a0ec7b5bfb03202be8b (this is just before #7365) got merged i get this error:

```
++ MbedOS Error Info ++
Error Status: 0x80FF0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x8000CC1
Error Value: 0x0
Current Thread: Id: 0x20001724 Entry: 0x8001741 StackSize: 0x1000 StackMem: 0x20001770 SP: 0x20002668
-- MbedOS Error Info --
rtc_synchronize error
```

### Issue request type
[ ] Question  
[ ] Enhancement  
[X] Bug

jeromecoutant, 2018-07-17:
Hi

About the 2 reported issues:

1. "RTC initialization failed" : this is the same as #7497. I will try again to reproduce.

2. "rtc_synchronize error" : if you go back with old code, you need to unplug your board after downloading the bin.
This is because RTC init procedure has been changed in #7365.

bcostm, 2018-07-17:
Just to confirm: are you using a NUCLEO_F412ZG or a custom board ?

pilotak, 2018-07-17:
custom board with RTC crystal LFXTAL009678 and main ABM3B-8.000MHZ-10  `mbed_app.json`

```json
    "target_overrides": {
        "NUCLEO_F412ZG": {
            "platform.stdio-convert-newlines": true,
            "platform.stdio-baud-rate": 115200,
            "target.clock_source": "USE_PLL_HSE_XTAL",
            "target.lse_available": 1,
            "target.stdio_uart_tx": "PA_9", 
            "target.stdio_uart_rx": "PA_10"
        }
    }
```

jeromecoutant, 2018-08-06:
Hi
Should we keep this PR opened ?

pilotak, 2018-08-06:
It's still happening

jeromecoutant, 2018-08-06:
OK

You need to help me to reproduce the issue.

pilotak, 2018-08-06:
Please refer to first post, nothing changed since then. I have now bootloader in place and it starts fine and when it finishes and actual app starts i then get an error.

jeromecoutant, 2018-08-08:
Your code is executing well...

set time
alive
uptime: 1
alive
uptime: 2
alive
uptime: 3

jeromecoutant, 2018-08-09:
Code is not backward compatible. You can't move from different git version.

jeromecoutant, 2018-08-09:
Maybe you can direct call rtc_init() function, and then add some wait before starting other RTC actions ?

But again,  the issue mentioned in the description "RTC initialization failed" is no more in the code...
So, I think we can close the issue.

Thx

pilotak, 2018-08-09:
` rtc_init()` didn't help, but thank you for your time

closed 2018-08-09 by pilotak