ARMmbed/mbed-os #7522: K64 mbed_hal-rtc_reset test timeout

Alex-EEE, 2018-07-16:
### Description

We run mbed test nightly against a K64 board and this test seems to timeout instead of passing about 70% of the time:  Do we need to increase a timeout or something?

018-07-16T14:09:21.7366534Z mbedgt: ['mbedhtrun', '-m', u'K64F', '-p', u'COM3:9600', '-f', u'"BUILD/tests/k64f/mbed/mbed-os/TESTS/mbed_hal/rtc_reset/rtc_reset.bin"', '-e', u'"mbed-os\\TESTS\\host_tests"', '-d', 'F:', '-c', 'jlink', '-t', u'0240000036934e45003840149f3500430e61000097969900', '-r', 'jlink', '-C', '4', '--sync', '5'] failed after 1 count
2018-07-16T14:09:21.7421429Z mbedgt: checking for GCOV data...
2018-07-16T14:09:21.7421429Z mbedgt: test case summary event not found
2018-07-16T14:09:21.7426422Z               no test case report present, assuming test suite to be a single test case!
2018-07-16T14:09:21.7431409Z               test suite: mbed-mbed-os-tests-mbed_hal-rtc_reset
2018-07-16T14:09:21.7436626Z               test case: mbed-mbed-os-tests-mbed_hal-rtc_reset
2018-07-16T14:09:21.7441378Z mbedgt: test on hardware with target id: 0240000036934e45003840149f3500430e61000097969900
2018-07-16T14:09:21.7446377Z mbedgt: test suite 'mbed-mbed-os-tests-mbed_hal-rtc_reset' ........................................... TIMEOUT in 114.08 sec
2018-07-16T14:09:21.7451364Z               test case: 'mbed-mbed-os-tests-mbed_hal-rtc_reset' ........................................... ERROR in 114.08 sec

### Issue request type

[ ] Question  
[ ] Enhancement  
[ X ] Bug

0xc0170, 2018-07-17:
@ARMmbed/team-nxp @mprse  Please review

mprse, 2018-07-17:
This is my output on `K64F/GCC_ARM`:
```
$ mbed test -t GCC_ARM -m K64F -n tests-mbed_hal-rtc_reset -v
Scan: .
Scan: FEATURE_LWIP
Scan: FEATURE_STORAGE
Scan: TESTS
Scan: TESTS
Scan: TESTS
Scan: TESTS
Scan: TESTS
Scan: TESTS
Scan: TESTS
Scan: TESTS
Scan: TESTS
Building library mbed-build (K64F, GCC_ARM)
Scan: mbed-os
Scan: FEATURE_LWIP
Scan: FEATURE_STORAGE
Macros: -DDEVICE_EMAC=1 -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DTARGET_Freescale -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DMBED_BUILD_TIMESTAMP=1531821913.75 -DDEVICE_RTC=1 -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DFSL_RTOS_MBED -DDEVICE_STORAGE=1 -DTARGET_KPSDK_MCUS -DTOOLCHAIN_GCC -DDEVICE_CRC=1 -DTARGET_CORTEX_M -DTARGET_KSDK2_MCUS -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_K64F -DDEVICE_SPI_ASYNCH=1 -DDEVICE_LPTICKER=1 -DDEVICE_PWMOUT=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_Freescale_EMAC -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DCPU_MK64FN1M0VMD12 -DTARGET_LIKE_MBED -DTARGET_FF_ARDUINO -DTARGET_KPSDK_CODE -DTARGET_RELEASE -DDEVICE_SERIAL_FC=1 -DFEATURE_STORAGE=1 -DDEVICE_USTICKER=1 -DDEVICE_TRNG=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DTARGET_FRDM -DTARGET_MCUXpresso_MCUS -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DDEVICE_FLASH=1 -DDEVICE_PORTIN=1 -DTARGET_MCU_K64F -DARM_MATH_CM4
Scan: GCC_ARM
Scan: FEATURE_LWIP
Scan: FEATURE_STORAGE
Building project rtc_reset (K64F, GCC_ARM)
Scan: GCC_ARM
Scan: rtc_reset
Scan: FEATURE_LWIP
Scan: FEATURE_STORAGE
Macros: -DMBED_BUILD_TIMESTAMP=1531821916.93 -DDEVICE_EMAC=1 -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DTARGET_Freescale -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DDEVICE_RTC=1 -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DFSL_RTOS_MBED -DDEVICE_STORAGE=1 -DTARGET_KPSDK_MCUS -DTOOLCHAIN_GCC -DDEVICE_CRC=1 -DTARGET_CORTEX_M -DTARGET_KSDK2_MCUS -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_K64F -DDEVICE_SPI_ASYNCH=1 -DDEVICE_LPTICKER=1 -DDEVICE_PWMOUT=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_Freescale_EMAC -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DCPU_MK64FN1M0VMD12 -DTARGET_LIKE_MBED -DTARGET_FF_ARDUINO -DTARGET_KPSDK_CODE -DTARGET_RELEASE -DDEVICE_SERIAL_FC=1 -DFEATURE_STORAGE=1 -DDEVICE_USTICKER=1 -DDEVICE_TRNG=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DTARGET_FRDM -DTARGET_MCUXpresso_MCUS -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DDEVICE_FLASH=1 -DDEVICE_PORTIN=1 -DTARGET_MCU_K64F -DARM_MATH_CM4
Link: rtc_reset
Preproc: C:\Programs\Arm-none-aebi\6 2017-q2-update\bin\arm-none-eabi-cpp -E -P BUILD\tests\K64F\GCC_ARM\targets\TARGET_Freescale\TARGET_MCUXpresso_MCUS\TARGET_MCU_K64F\device\TOOLCHAIN_GCC_ARM\MK64FN1M0xxx12.ld -Wl,--gc-sections -Wl,--wrap,main -Wl,--wrap,_malloc_r -Wl,--wrap,_free_r -Wl,--wrap,_realloc_r -Wl,--wrap,_memalign_r -Wl,--wrap,_calloc_r -Wl,--wrap,exit -Wl,--wrap,atexit -Wl,-n -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -o C:\Projects\mbed-os\BUILD\tests\K64F\GCC_ARM\.\TESTS\mbed_hal\rtc_reset\.link_script.ld
[DEBUG] Return: 0
Link: C:\Programs\Arm-none-aebi\6 2017-q2-update\bin\arm-none-eabi-gcc @C:\Projects\mbed-os\BUILD\tests\K64F\GCC_ARM\.\TESTS\mbed_hal\rtc_reset\.link_files.txt
[DEBUG] Return: 0
Elf2Bin: rtc_reset
FromELF: C:\Programs\Arm-none-aebi\6 2017-q2-update\bin\arm-none-eabi-objcopy -O binary C:\Projects\mbed-os\BUILD\tests\K64F\GCC_ARM\.\TESTS\mbed_hal\rtc_reset\rtc_reset.elf C:\Projects\mbed-os\BUILD\tests\K64F\GCC_ARM\.\TESTS\mbed_hal\rtc_reset\rtc_reset.bin
[DEBUG] Return: 0
+---------------------------------+-------+-------+------+
| Module                          | .text | .data | .bss |
+---------------------------------+-------+-------+------+
| TESTS\mbed_hal                  |   365 |     0 |  204 |
| [fill]                          |   121 |     0 | 2403 |
| [lib]\c.a                       | 36935 |  2472 |   89 |
| [lib]\gcc.a                     |  3392 |     0 |    0 |
| [lib]\misc                      |   248 |     8 |   28 |
| [lib]\nosys.a                   |    32 |     0 |    0 |
| drivers\InterruptIn.o           |    52 |     0 |    0 |
| drivers\RawSerial.o             |   148 |     0 |    0 |
| drivers\SPI.o                   |    16 |     0 |    0 |
| drivers\Serial.o                |    36 |     0 |    0 |
| drivers\SerialBase.o            |   689 |     0 |    0 |
| drivers\UARTSerial.o            |    20 |     0 |    0 |
| features\frameworks             |  1449 |    36 |  312 |
| features\storage                |    42 |     0 |  184 |
| hal\mbed_critical_section_api.o |   166 |     0 |    2 |
| hal\mbed_gpio.o                 |    96 |     0 |    0 |
| hal\mbed_pinmap_common.o        |   260 |     0 |    0 |
| hal\mbed_sleep_manager.o        |   231 |     0 |    2 |
| hal\mbed_ticker_api.o           |   872 |     0 |    0 |
| hal\mbed_us_ticker_api.o        |    64 |     4 |   64 |
| platform\FileHandle.o           |     4 |     0 |    0 |
| platform\mbed_alloc_wrappers.o  |    24 |     0 |    0 |
| platform\mbed_assert.o          |    85 |     0 |    0 |
| platform\mbed_board.o           |   204 |     0 |    0 |
| platform\mbed_critical.o        |   230 |     0 |    4 |
| platform\mbed_error.o           |   814 |     0 |  113 |
[... cut ...] 00430e61000097969900', '-r', 'jlink', '-C', '4', '--sync', '5'] failed after 1 count
```
It looks like you are not using the default setting for flashing the target and resetting the target (`'-c', 'jlink','-r', 'jlink'`). Maybe this is causing the problem and sometimes `mbedhtrun` fails.

This test is a two part test which runs on host and the device. The test resets the device so maybe this is some kind of host-device communication issue. 

@0xc0170 Do you noticed such problems on morph tests?

0xc0170, 2018-07-17:
> @0xc0170 Do you noticed such problems on morph tests?

No, we haven't .

Alex-EEE, 2018-07-17:
We're using a custom written plugin to flash with Seggar j-link instead of DAPLink.  We'll take a look at the reset code.  The problem is probably on our end if it's working for yall.  Thanks for the look!

closed 2018-07-17 by Alex-EEE