ARMmbed/mbed-os #7526: Building failed on Arm CM3DS board

davidhuziji, 2018-07-17:
Hi,

I'm trying to run [Blinky example](https://github.com/ARMmbed/mbed-os-example-blinky) on Arm CM3DS board.
After downloading the code, I input command `mbed compile -v -t ARM -m ARM_CM3DS_MPS2` to build the example as the README requires. But an errors occurred during building.

The error log shows as below:
```
XXX@YYY:/.../arm-mbed/mbed-os-example-blinky$ mbed compile -v -t ARM -m ARM_CM3DS_MPS2
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "/.../arm-mbed/mbed-os-example-blinky" (program)
[mbed] Exec "/usr/bin/python3 -u /.../arm-mbed/mbed-os-example-blinky/mbed-os/tools/make.py -t ARM -m ARM_CM3DS_MPS2 --source . --build ./BUILD/ARM_CM3DS_MPS2/ARM -v" in "/.../arm-mbed/mbed-os-example-blinky"
Traceback (most recent call last):
  File "/.../arm-mbed/mbed-os-example-blinky/mbed-os/tools/make.py", line 41, in <module>
    from tools.tests import TESTS, Test, TEST_MAP
  File "/.../arm-mbed/mbed-os-example-blinky/mbed-os/tools/tests.py", line 18, in <module>
    from tools.data.support import DEFAULT_SUPPORT, CORTEX_ARM_SUPPORT
  File "/.../arm-mbed/mbed-os-example-blinky/mbed-os/tools/data/support.py", line 17, in <module>
    from tools.targets import TARGETS
  File "/.../arm-mbed/mbed-os-example-blinky/mbed-os/tools/targets/__init__.py", line 557, in <module>
    update_target_data()
  File "/.../arm-mbed/mbed-os-example-blinky/mbed-os/tools/targets/__init__.py", line 546, in update_target_data
    in Target.get_json_target_data().items()
  File "/.../arm-mbed/mbed-os-example-blinky/mbed-os/tools/targets/__init__.py", line 70, in wrapper
    CACHES[(func.__name__, args)] = func(*args, **kwargs)
  File "/.../arm-mbed/mbed-os-example-blinky/mbed-os/tools/targets/__init__.py", line 142, in get_json_target_data
    Target.__targets_json_location_default)
  File "/.../arm-mbed/mbed-os-example-blinky/mbed-os/tools/utils.py", line 351, in json_file_to_dict
    object_pairs_hook=OrderedDict)
  File "/usr/lib/python3.5/json/__init__.py", line 312, in loads
    s.__class__.__name__))
TypeError: the JSON object must be str, not 'bytes'
[mbed] ERROR: "/usr/bin/python3" returned error.
       Code: 1
       Path: "/.../arm-mbed/mbed-os-example-blinky"
       Command: "/usr/bin/python3 -u /.../arm-mbed/mbed-os-example-blinky/mbed-os/tools/make.py -t ARM -m ARM_CM3DS_MPS2 --source . --build ./BUILD/ARM_CM3DS_MPS2/ARM -v"
       Tip: You could retry the last command with "-v" flag for verbose output
```

The mbed OS version is 5.9
`XXX@YYY:/.../arm-mbed/mbed-os-example-blinky$ mbed ls
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
mbed-os (#62f8b922b420, tags: latest, mbed-os-5.9.2)
`
I tried GCC ARM and ARM Compiler. But both failed with the same error log.
Both python2.7 and python3 are installed on my Ubuntu. I use an alias to re-direct python and python2 to python2.7. But it seems that it doesn't help.

Could you please help check my issue? Thanks a lot in advance.

[ ] Question  
[ ] Enhancement  
[X ] Bug

davidhuziji, 2018-07-17:
mbed version: 1.7.5

0xc0170, 2018-07-17:
`ERROR: "/usr/bin/python3" returned error.` from the error it seems this is python 3 error

What is the version of mbed-cli ? there was a recent bugfix that should resolve this.

davidhuziji, 2018-07-17:
@0xc0170 , thanks a lot for the reply. Could you tell me how to get the mbed-cli version?

0xc0170, 2018-07-17:
`mbed --version`

I am currently on 1.7.3 , python2, no errors using latest master or latest 5.9.3 release

davidhuziji, 2018-07-17:
Mine is 1.7.5

0xc0170, 2018-07-17:
Thanks for the info. We already have this reported https://github.com/ARMmbed/mbed-os/issues/7026. 

I'll close this as duplicate (already referenced the original issue).

closed 2018-07-17 by 0xc0170

davidhuziji, 2018-07-18:
@0xc0170 Thanks. Maybe a irrelevant question. May I know how to force mbed to use Python 2 in Ubuntu which uses Python 3 by default? Using Python virtualenv or any other more easy way?