ARMmbed/mbed-os #7535: test: some cases are failing with GCC_ARM for Toshiba's TMPM3H6

ganesh-ramachandran, 2018-07-18:
### Description

**Target:**
Toshiba's TMPM3H6

**TMPM3H6 Memory:**
RAM - 16K and Flash - 128K

**ToolChain:**
GCC_ARM

**Issue:**
Following test cases failed on TMPM3H6 with GCC_ARM - failed for memory allocation on each case. ARM or IAR had passed with no error.
- features-tests-filesystem-buffered_block_device
- features-tests-filesystem-heap_block_device
- features-tests-filesystem-mbr_block_device
- features-tests-filesystem-util_block_device
- tests-mbedmicro-rtos-mbed-threads

*Note:* If main stack memory is reduced to 2K for GCC_ARM in [mbed_rtx.h](https://github.com/ARMmbed/mbed-os/files/2204358/mbed_rtx.txt), then Greentea tests are passed.

**Expected behavior**
Above mentioned Greentea test cases should pass without change in main stack.

**Actual behavior**
Above mentioned Greentea test cases fails due to not enough memory (Operator new[] out of memory).

**Steps to reproduce**
1. Download master from https://github.com/ganesh-ramachandran/mbed-os (since Toshiba TMPM3H6 is yet to be merged)

2. Build command
```
mbed test -t GCC_ARM -m TMPM3H6 -v -n features-tests-filesystem-buffered_block_device,features-tests-filesystem-heap_block_device,features-tests-filesystem-mbr_block_device,features-tests-filesystem-util_block_device,tests-mbedmicro-rtos-mbed-threads
```

**Logs**
Attached below the GCC_ARM logs with 4K and 2K main stack size.
[M3H6_GCC_ARM_GreenTea_MainStack_4K.txt](https://github.com/ARMmbed/mbed-os/files/2204716/M3H6_GCC_ARM_GreenTea_MainStack_4K.txt)
[M3H6_GCC_ARM_GreenTea_MainStack_2K.txt](https://github.com/ARMmbed/mbed-os/files/2204717/M3H6_GCC_ARM_GreenTea_MainStack_2K.txt)

**Related Issues**
Related issues are in #7115 and #4901

### Issue request type

[ ] Question
[ ] Enhancement
[X] Bug

davidsaada, 2018-07-18:
@ganesh-ramachandran Can you check if [#7465](https://github.com/ARMmbed/mbed-os/pull/7465) resolves the first 4 tests? 5th test should be handled separately.

ganesh-ramachandran, 2018-07-19:
> @ganesh-ramachandran Can you check if #7465 resolves the first 4 tests? 5th test should be handled separately.

@davidsaada, Thank you for your inputs. By referring #7465, the first 4 tests are resolved.

For 5th test case, observed that test case is passed by reducing '[CHILD_THREAD_STACK_SIZE](https://github.com/ARMmbed/mbed-os/blob/master/TESTS/mbedmicro-rtos-mbed/threads/main.cpp#L40)' from 384 to 223. Please confirm.

davidsaada, 2018-07-19:
Thanks @ganesh-ramachandran. Gonna apply a similar fix as in [#7465](https://github.com/ARMmbed/mbed-os/pull/7465) to tests-mbedmicro-rtos-mbed-threads as well. Either in the same PR or in a separate one.

0xc0170, 2018-08-07:
>Thanks @ganesh-ramachandran. Gonna apply a similar fix as in #7465 to tests-mbedmicro-rtos-mbed-threads as well. Either in the same PR or in a separate one.

@davidsaada Can you send the PR to fix also threads test?

davidsaada, 2018-08-07:
> @davidsaada Can you send the PR to fix also threads test?

It's on my todo list, but will take time to resolve (haven't started working on it yet).

davidsaada, 2018-08-08:
@ganesh-ramachandran As for the tests-mbedmicro-rtos-mbed-threads test: Do you build it with `-DMBED_STACK_STATS_ENABLED=1` flag? Alternatively, does this fail as part of our CI run?
If any of these is true, can you check whether [#7730](https://github.com/ARMmbed/mbed-os/pull/7730) resolves this problem?

davidsaada, 2018-08-09:
@ganesh-ramachandran Please check whether [#7744](https://github.com/ARMmbed/mbed-os/pull/7744) solves the `tests-mbedmicro-rtos-mbed-threads` test problem.

closed 2018-08-23 by cmonr