ARMmbed/mbed-os #7537: Greentea key-value pairs are messed up by other threads.

SierpG, 2018-07-18:
### Description

I am writing tests for a mbed application using greentea. In my test suits it is necessary to have several threads. My problem is the following:

Thread 1 sends with greentea_send_kv a key-value pair and waits with greentea_parse_kv for the result of the host test.
At the same time thread 2 writes with debug() or utest_printf some information to the UART. 
If this happens at the same time as thread 1 uses greentea_send_kv, it happens that the keyword is messed up with some characters from the message of thread 2 and the hosts-test cannot recognize the keyword. Then thread 1 waits for ever in greentea_parse_kv and the test fails with timeout.

I solved this problem with a mutex, which every thread has to lock before any of the critical functions are used. However this is only possible within the scope of my test. If for example greentea itself sends keywords like __testcase_start or __testcase_finish this cannot be directly controlled. 

### Issue request type

[X ] Question  
[ ] Enhancement  
[ ] Bug  

Are there any posibilities provided by greentea to synchronize the output to the UART from different threads or are there plans to implement something like that? Using a second UART is no option for me, because not all the hardware I am testing on, has a second UART. Implementing the synchronisiation in the test cases itself is a quite big overhead. Doing the synchronisation at a lower lever whould be preferable.

jamesbeyond, 2018-07-24:
Hi @SierpG,

Currently, mbed greentea is designed to work with one UART port only by default. and it is not threadsafe unless you use a mutex.
Could you explain your case a bit more, why do you need the second UART port within Greentea test.
If there is a valid use case, we will try to come up with some solutions.

SierpG, 2018-07-25:
I do not need a second UART. 
The problem is that if several threads write to the UART at the same time, the output gets messed up, because at the end putc is used. For example sending a key word with greentea_send_kv does:
        greentea_write_preamble();
        greentea_write_string(key);
        greentea_serial->putc(';');
        greentea_write_string(val);
        greentea_write_postamble();
If another thread makes output to the UART at the same time, the key word-value pair is messed up.

Ofcourse one can use a mutex in the test cases. But our software uses also debug() from mbed_debug, which goes directly to the low level functions, which write to the UART. This means for us to write a new interface over mbed_debug.

0xc0170, 2018-07-27:
`debug()` from `mbed_debug.h` ? That uses stdout default console.

One way to fix  this is via `mbed_override_console`, see https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_retarget.h#L133

As greentea serial is implemented it uses RawSerial. Therefore what you can do in your app is to override the console to use green tea serial object (use greentea serial object for stdin/out/err) - use one "serial" object. This however does not fix thread as RawSerial is not threadsafe by default (`GreenteaSerial` could implement lock/unlock) - this needs to be implemented.

@jamesbeyond Please review

jamesbeyond, 2018-07-30:
Hi @SierpG,

As @0xc0170 suggested, we don't have a ready solution for your case of threadsafe Greentea RawSerial output. This is one of the GreenTea limitations, unfortunately.

When we design the tool we assume that it uses the UART communication channel exclusively. We don't expect other threads use the UART arbitrarily. At least when we design our test cases, we are following this rule. 

In your app, If the `debug()` output, if they are not essential to the test, maybe they can be turned off via defining `NDEBUG`? Or move to the same thread with the Greentea test, and use utest_printf() instead?

0xc0170, 2018-08-02:
Thanks @SierpG for the details.

@jamesbeyond Shall this be closed as answered?

SierpG, 2018-08-02:
Yes

closed 2018-08-02 by 0xc0170