ARMmbed/mbed-os #7538: mbed-os-example-blinky does not build on mbed IDE

bcostm, 2018-07-18:
### Description

mbed-os revision: 50bd61a

Error: Symbol __user_perthread_libspace multiply defined (by /build/mbed-os.lib/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_lib.NUCLEO_F412ZG.o and /build/mbed-os.lib/rtos/TARGET_CORTEX/mbed_boot.NUCLEO_F412ZG.o).
Error: Not enough information to list the image map.
Info: Unable to download. Fix the reported errors...

Tested with NUCLEO_F412ZG and DISCO_L496AG

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-07-18:
5.9.3 should have fixed it. I've just reproduced this, we will investigate

@theotherjimmy

bcostm, 2018-07-19:
I didn't it was already mentioned in Issue #7422

Another workaround for the moment, is to add the _RTE_ macro using [Compile] > [Compile macros]  in the mbed IDE.

cmonr, 2018-07-19:
@bcostm This should now be fixed with the 5.9.3 release. Please confirm on your end

bcostm, 2018-07-20:
On the mbed IDE, the mbed library is the version 162 which corresponds to mbed-os-5.9.1. So the issue is still present for the moment.

bcostm, 2018-07-20:
Yes I confirm the mbed-os-example-blinky compiled OK now.
Thanks

closed 2018-07-20 by bcostm