ARMmbed/mbed-os #7541: UBLOX EVK-NINA-B1 (nRF52832) BLE_HeartRate example not working

edablan, 2018-07-18:
### Description

I cloned the mbed-os-example-ble repo, then deployed the mbed-os inside BLE_HeartRate example. I compiled both in the mbed-cli using the IAR compiler, as well as exporting to IAR IDE to build and saw the same behavior. The EVK-NINA-B1 will advertise with the correct name "HRM", but when trying to connect using an iOS, android, or windows device it fails to connect. I continue to retry to connect, but eventually the EVK-NINA-B1 stops advertising.

    - EVK-NINA-B1
    - IAR version (7.80.4)
    - mbed-cli (version 1.3.0)
    - mbed-os (65afa82)

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-07-19:
cc @ARMmbed/team-ublox

edablan, 2018-07-20:
Ok so I think I have noticed some things. All of the BLE_EXAMPLES seem to work on the EVK-NINA-B1 as long as you follow the below steps, and you don't run it in debug mode of IAR. For whatever reason even using an external J-Link Base debugger through SWD makes it impossible to connect to a BLE device while in debug mode in IAR. Not sure if this is related to [https://github.com/ARMmbed/mbed-os-example-ble/issues/39](url)
Though this is on a different board, and the EVK-NINA-B1 does not use CMSIS-DAP.

1. Erase chip completely, then Flash Soft Device using nRFgo Studio
2. Use the Linker Script fix issued a few days ago for any nRF52832 (seems this will be fixed by 5.9.4?) #7533 
3. In IAR go to Project->Download->Download Active Application
4. Physically power cycle the EVK-NINA-B1 (plug and unplug the power source manually). Hitting the reset button doesn't do it (at least from my attempts)

Is this procedure intended or is there something going on here?

marcuschangarm, 2018-07-23:
I don't have IAR IDE, but with DAPLINK, NRF52_DK, and IAR commandline compiler I can run the heart rate example with GDB debugger connected.

Have you tried the newest DAPLINK IF firmware? 
https://armmbed.github.io/DAPLink/

edablan, 2018-07-23:
We will need to debug using IAR, since this is the compiler we are building against. We want the features of IAR, and were told it is supported by mbed. I also tried this with the I-Jet debugger and seeing the same behavior.

marcuschangarm, 2018-07-23:
DAPLINK is compatible with IAR. I'm suspecting the problems you are seeing are related to the interface firmware and updating it might help.

edablan, 2018-07-24:
DAPLINK isn't working out of the box with the EVK-NINA-B1. Do we need to load a new bootloader for this to work with mbed?

edablan, 2018-07-24:
Also the EVK-NINA-B1 does not show up as a CMSIS-DAP device, it shows up as a J-Link device. It does not use DAP Link, but uses a built-in J-Link debugger. It is also setup to use external debuggers through SWD. It's currently failed to debug with two different external debuggers( J-Link, and I-jet) as well as failing with the built in J-link debugger.

edablan, 2018-07-27:
Ok seems to be working now in debug mode on two different laptops using J-Link and I-jet. Not sure what happened. Going to close this out then.

edablan, 2018-08-01:
@marcuschangarm 

So I am noticing this as an intermittent issue on the EVK-NINA-B1 board, and having issues knowing what puts it into the following state. I am using the built in debugger of the EVK-NINA-B1, which is not CMSIS-DAP but a J-Link debugger. When in debug mode (and presumably non debug mode) the line inside of the BLE_HearRate main.cpp file:

```C++
    eventQueue.dispatch_forever();
```
is causing hard faults. Though there are rare times where this goes away on its own without me changing anything in debug mode, and it will let the heart rate example work (or any BLE example). Even if I simply flash the device in IAR in non-debug mode, the heart rate example will not advertise (I'm assuming the hard fault issue is causing this). Though if I manually power cycle the EVK-NINA-B1 several times it will all of a sudden start advertising and connect. FYI i'm not just seeing this behavior with the BLE_HeartRate example but the others. Has anyone actually tried using the EVK-NINA-B1 board with IAR, and not the NRF52_DK?

closed 2018-09-25 by edablan