ARMmbed/mbed-os #7542: NUMAKER_PFM_M2351 serial communication is broken on linux.

studavekar, 2018-07-18:
### Description

NUMAKER_PFM_M2351 serial communication is broken on linux. It works well on windows 10 

** Host machine info**
lsb_release -a

```
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.2 LTS
Release:        16.04
Codename:       xenial
```
uname -a 
`Linux austin-ci-linux-002 4.15.0-24-generic #26~16.04.1-Ubuntu SMP Fri Jun 15 14:35:08 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux`

NUMAKER_PFM_M2351

**Toolchain:**
ARMC6

**mbed-os sha:**
e9e1ff9 Merge pull request #7302 from OpenNuvoton/nuvoton_m2351_v1.1

**Steps to reproduce**
Use blinky program with uart logging, expect to see led blink and incremental counter value on uart.

```
#include "mbed.h"

DigitalOut led1(LED1);
int main() {
    uint32_t counter = 0;
    while (1) {
        led1=!led1;
        wait(0.5);
        ++counter;
        printf("counter : %d \n\r",counter);
    }
}
```

**Logs**

Tool used : Beagle USB 480 Protocol Analyzer , software is availabe for free https://www.totalphase.com/products/beagle-usb480/ 

Please rename the files from `.txt` to `.tdc` to view it 

- [linux_failure.tdc.txt](https://github.com/ARMmbed/mbed-os/files/2206804/linux_failure.tdc.txt)

- [windows_success.tdc.txt](https://github.com/ARMmbed/mbed-os/files/2206805/windows_success.tdc.txt)

[ ] Question  
[ ] Enhancement  
[ X] Bug  

cc : @deepikabhavnani @SenRamakri @c1728p9 @ccli8

deepikabhavnani, 2018-07-18:
CC @cyliangtw @ccli8

ccli8, 2018-07-24:
@studavekar @deepikabhavnani We can now reproduce on Ubuntu 16.04. Our tool team is looking into it.

samchuarm, 2018-08-06:
@studavekar @deepikabhavnani Has this issue been verified with the latest M2351 interface firmware binary provided by Nuvoton Chris? Thanks.

studavekar, 2018-08-06:
@samchuarm this is been fixed in the latest interface binary provided.

Thanks @ccli8.

closed 2018-08-06 by studavekar