ARMmbed/mbed-os #7550: Bug in `test` directory - linker flag path

OmerRosler, 2018-07-19:
Hello

I was trying to run `mbed test` on my project, however it seems that there is a bug there.

The `--map` flag get passed to the linker without quotation marks (my project path contains spaces) and that causes the command to crash.

It originates from the file `PROJECT\TESTS\mbed_hal\ticker\.link_files.txt` or any other `.link_files.txt` files.

I think the bug originates from the `tools` directory and not the cli therefore I post it here.

theotherjimmy, 2018-09-24:
@OmerRosler I think this should be resolved now that `--build` is not passed as an absolute path.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-272*

adbridge, 2020-10-13:
We've updated our automation, I will fix the requirements .