ARMmbed/mbed-os #7555: NVStore tests hard fault for MOTE_L152RC

bentcooke, 2018-07-19:
### Description
NVStore tests hard fault for MOTE_L152RC target.  This appears to be related to the comments in #7127.  

$ mbed test -t GCC_ARM -m MOTE_L152RC -n mbed-os-features-nvstore* -vv

```
[1532013036.54][CONN][RXD] >>> Running case #1: 'NVStore: Basic functionality'...
[1532013036.59][CONN][INF] found KV pair in stream: {{__testcase_start;NVStore: Basic functionality}}, queued...
[1532013036.60][CONN][RXD] NVStore areas:
[1532013036.90][CONN][RXD] Area 0: address 0x0803e000, size 4096 (0x1000)
[1532013036.95][CONN][RXD] Area 1: address 0x0803f000, size 4096 (0x1000)
[1532013037.48][CONN][RXD] Max keys 20 (out of 255 possible ones)
[1532013037.91][CONN][RXD]
[1532013037.94][CONN][RXD] ++ MbedOS Fault Handler ++
[1532013037.94][CONN][RXD]
[1532013037.96][CONN][RXD] FaultType: HardFault
[1532013037.96][CONN][RXD]
[1532013037.97][CONN][RXD] Context:
[1532013037.99][CONN][RXD] R0   : 20001FF8
[1532013038.00][CONN][RXD] R1   : 3803E0F8
[1532013038.02][CONN][RXD] R2   : FFFFFFE8
[1532013038.04][CONN][RXD] R3   : 00000000
[1532013038.05][CONN][RXD] R4   : 3803E0F8
[1532013038.07][CONN][RXD] R5   : 20001FF8
[1532013038.09][CONN][RXD] R6   : 00000008
[1532013038.10][CONN][RXD] R7   : 20002980
[1532013038.13][CONN][RXD] R8   : 00000000
[1532013038.14][CONN][RXD] R9   : 20001FF8
[1532013038.16][CONN][RXD] R10  : 000000BF
[1532013038.18][CONN][RXD] R11  : 00000017
[1532013038.19][CONN][RXD] R12  : 80000000
[1532013038.21][CONN][RXD] SP   : 20001FC0
[1532013038.23][CONN][RXD] LR   : 0800B5AF
[1532013038.24][CONN][RXD] PC   : 080015C4
[1532013038.26][CONN][RXD] xPSR : 81004400
[1532013038.27][CONN][RXD] PSP  : 20001FA0
[1532013038.29][CONN][RXD] MSP  : 20007FD8
[1532013038.31][CONN][RXD] CPUID: 412FC230
[1532013038.32][CONN][RXD] HFSR : 40000000
[1532013038.34][CONN][RXD] MMFSR: 00000000
[1532013038.36][CONN][RXD] BFSR : 00000082
[1532013038.37][CONN][RXD] UFSR : 00000000
[1532013038.39][CONN][RXD] DFSR : 00000008
[1532013038.40][CONN][RXD] AFSR : 00000000
[1532013038.42][CONN][RXD] BFAR : 3803E0F8
[1532013038.43][CONN][RXD] Mode : Thread
[1532013038.45][CONN][RXD] Priv : Privileged
[1532013038.46][CONN][RXD] Stack: PSP
[1532013038.46][CONN][RXD]
[1532013038.49][CONN][RXD] -- MbedOS Fault Handler --
[1532013038.49][CONN][RXD]
[1532013038.49][CONN][RXD]
[1532013038.51][CONN][RXD]
[1532013038.53][CONN][RXD] ++ MbedOS Error Info ++
[1532013038.58][CONN][RXD] Error Status: 0x80FF013D Code: 317 Module: 255
[1532013038.62][CONN][RXD] Error Message: Unrecoverable fault excaption.
[1532013038.64][CONN][RXD] Location: 0x800BC3F
[1532013038.67][CONN][RXD] Error Value: 0x80015C4
[1532013038.77][CONN][RXD] Current Thread: Id: 0x20002298 Entry: 0x800D859 StackSize: 0x1000 StackMem: 0x20001298 SP: 0x20007F68
[1532013038.80][CONN][RXD] -- MbedOS Error Info --
```

### Issue request type
[ ] Question  
[ ] Enhancement  
[ x] Bug  

@davidsaada

dannybenor, 2018-07-23:
@bentcooke  did you use the mbedos code after https://github.com/ARMmbed/mbed-os/pull/7127 was merged?

bentcooke, 2018-07-23:
@dannybenor Yes, I am using the fixes included in #7127, but the hard fault still occurs with this platform.

davidsaada, 2018-08-08:
@bentcooke Please check whether [#7730](https://github.com/ARMmbed/mbed-os/pull/7730) resolves this issue.

dannybenor, 2018-08-22:
@bentcooke Can you close this issue?

bentcooke, 2018-08-22:
This is now fixed with #7746

closed 2018-08-22 by bentcooke