ARMmbed/mbed-os #7557: lp ticker speed test failure for MOTE_L152RC target

bentcooke, 2018-07-19:
### Description
The "lp ticker speed test" fails for the MOTE_L152RC for all three compilers.  I have not tested the other STM32L15x platforms but I expect they will have the same issue.

$ mbed test -t GCC_ARM -m MOTE_L152RC -n mbed-os-tests-mbed_hal-common_tickers -vv

```
mbedgt: test case report:
+---------------------+---------------+---------------------------------------+----------------------------------------------------+--------+--------+--------+--------------------+
| target              | platform_name | test suite                            | test case                                          | passed | failed | result | elapsed_time (sec) |
+---------------------+---------------+---------------------------------------+----------------------------------------------------+--------+--------+--------+--------------------+
| MOTE_L152RC-GCC_ARM | MOTE_L152RC   | mbed-os-tests-mbed_hal-common_tickers | Microsecond ticker fire interrupt                  | 1      | 0      | OK     | 0.06               |
| MOTE_L152RC-GCC_ARM | MOTE_L152RC   | mbed-os-tests-mbed_hal-common_tickers | Microsecond ticker increment test                  | 1      | 0      | OK     | 0.22               |
| MOTE_L152RC-GCC_ARM | MOTE_L152RC   | mbed-os-tests-mbed_hal-common_tickers | Microsecond ticker info test                       | 1      | 0      | OK     | 0.05               |
| MOTE_L152RC-GCC_ARM | MOTE_L152RC   | mbed-os-tests-mbed_hal-common_tickers | Microsecond ticker init is safe to call repeatedly | 1      | 0      | OK     | 0.09               |
| MOTE_L152RC-GCC_ARM | MOTE_L152RC   | mbed-os-tests-mbed_hal-common_tickers | Microsecond ticker interrupt test                  | 1      | 0      | OK     | 0.06               |
| MOTE_L152RC-GCC_ARM | MOTE_L152RC   | mbed-os-tests-mbed_hal-common_tickers | Microsecond ticker overflow test                   | 1      | 0      | OK     | 0.09               |
| MOTE_L152RC-GCC_ARM | MOTE_L152RC   | mbed-os-tests-mbed_hal-common_tickers | Microsecond ticker past interrupt test             | 1      | 0      | OK     | 0.07               |
| MOTE_L152RC-GCC_ARM | MOTE_L152RC   | mbed-os-tests-mbed_hal-common_tickers | Microsecond ticker reschedule test                 | 1      | 0      | OK     | 0.06               |
| MOTE_L152RC-GCC_ARM | MOTE_L152RC   | mbed-os-tests-mbed_hal-common_tickers | Microsecond ticker speed test                      | 1      | 0      | OK     | 0.07               |
| MOTE_L152RC-GCC_ARM | MOTE_L152RC   | mbed-os-tests-mbed_hal-common_tickers | lp ticker fire interrupt                           | 1      | 0      | OK     | 0.06               |
| MOTE_L152RC-GCC_ARM | MOTE_L152RC   | mbed-os-tests-mbed_hal-common_tickers | lp ticker increment test                           | 1      | 0      | OK     | 0.05               |
| MOTE_L152RC-GCC_ARM | MOTE_L152RC   | mbed-os-tests-mbed_hal-common_tickers | lp ticker info test                                | 1      | 0      | OK     | 0.04               |
| MOTE_L152RC-GCC_ARM | MOTE_L152RC   | mbed-os-tests-mbed_hal-common_tickers | lp ticker init is safe to call repeatedly          | 1      | 0      | OK     | 0.2                |
| MOTE_L152RC-GCC_ARM | MOTE_L152RC   | mbed-os-tests-mbed_hal-common_tickers | lp ticker interrupt test                           | 1      | 0      | OK     | 0.32               |
| MOTE_L152RC-GCC_ARM | MOTE_L152RC   | mbed-os-tests-mbed_hal-common_tickers | lp ticker overflow test                            | 1      | 0      | OK     | 0.08               |
| MOTE_L152RC-GCC_ARM | MOTE_L152RC   | mbed-os-tests-mbed_hal-common_tickers | lp ticker past interrupt test                      | 1      | 0      | OK     | 0.05               |
| MOTE_L152RC-GCC_ARM | MOTE_L152RC   | mbed-os-tests-mbed_hal-common_tickers | lp ticker reschedule test                          | 1      | 0      | OK     | 0.25               |
| MOTE_L152RC-GCC_ARM | MOTE_L152RC   | mbed-os-tests-mbed_hal-common_tickers | lp ticker speed test                               | 0      | 1      | FAIL   | 0.15               |
+---------------------+---------------+---------------------------------------+----------------------------------------------------+--------+--------+--------+--------------------+
```

### Issue request type
[ ] Question  
[ ] Enhancement  
[x ] Bug

jeromecoutant, 2018-07-20:
Hi

Here is status with latest 5.9.3 release:

| target              | platform_name | test suite                            | test case                                          | passed | failed | result | elapsed_time (sec) |
|-|-|-|-|-|-|-|-|
| NUCLEO_L152RE-GCC_ARM | NUCLEO_L152RE | tests-mbed_hal-common_tickers                                                | Microsecond ticker fire interrupt                                                                 | 1      | 0      | OK      | 0.07               |
| NUCLEO_L152RE-GCC_ARM | NUCLEO_L152RE | tests-mbed_hal-common_tickers                                                | Microsecond ticker increment test                                                                 | 1      | 0      | OK      | 0.07               |
| NUCLEO_L152RE-GCC_ARM | NUCLEO_L152RE | tests-mbed_hal-common_tickers                                                | Microsecond ticker info test                                                                      | 1      | 0      | OK      | 0.06               |
| NUCLEO_L152RE-GCC_ARM | NUCLEO_L152RE | tests-mbed_hal-common_tickers                                                | Microsecond ticker init is safe to call repeatedly                                                | 1      | 0      | OK      | 0.09               |
| NUCLEO_L152RE-GCC_ARM | NUCLEO_L152RE | tests-mbed_hal-common_tickers                                                | Microsecond ticker interrupt test                                                                 | 1      | 0      | OK      | 0.06               |
| NUCLEO_L152RE-GCC_ARM | NUCLEO_L152RE | tests-mbed_hal-common_tickers                                                | Microsecond ticker overflow test                                                                  | 1      | 0      | OK      | 0.09               |
| NUCLEO_L152RE-GCC_ARM | NUCLEO_L152RE | tests-mbed_hal-common_tickers                                                | Microsecond ticker past interrupt test                                                            | 1      | 0      | OK      | 0.06               |
| NUCLEO_L152RE-GCC_ARM | NUCLEO_L152RE | tests-mbed_hal-common_tickers                                                | Microsecond ticker reschedule test                                                                | 1      | 0      | OK      | 0.07               |
| NUCLEO_L152RE-GCC_ARM | NUCLEO_L152RE | tests-mbed_hal-common_tickers                                                | Microsecond ticker speed test                                                                     | 1      | 0      | OK      | 0.06               |
| NUCLEO_L152RE-GCC_ARM | NUCLEO_L152RE | tests-mbed_hal-common_tickers                                                | lp ticker fire interrupt                                                                          | 1      | 0      | OK      | 0.04               |
| NUCLEO_L152RE-GCC_ARM | NUCLEO_L152RE | tests-mbed_hal-common_tickers                                                | lp ticker increment test                                                                          | 1      | 0      | OK      | 0.07               |
| NUCLEO_L152RE-GCC_ARM | NUCLEO_L152RE | tests-mbed_hal-common_tickers                                                | lp ticker info test                                                                               | 1      | 0      | OK      | 0.04               |
| NUCLEO_L152RE-GCC_ARM | NUCLEO_L152RE | tests-mbed_hal-common_tickers                                                | lp ticker init is safe to call repeatedly                                                         | 1      | 0      | OK      | 0.18               |
| NUCLEO_L152RE-GCC_ARM | NUCLEO_L152RE | tests-mbed_hal-common_tickers                                                | lp ticker interrupt test                                                                          | 1      | 0      | OK      | 0.31               |
| NUCLEO_L152RE-GCC_ARM | NUCLEO_L152RE | tests-mbed_hal-common_tickers                                                | lp ticker overflow test                                                                           | 1      | 0      | OK      | 0.08               |
| NUCLEO_L152RE-GCC_ARM | NUCLEO_L152RE | tests-mbed_hal-common_tickers                                                | lp ticker past interrupt test                                                                     | 1      | 0      | OK      | 0.06               |
| NUCLEO_L152RE-GCC_ARM | NUCLEO_L152RE | tests-mbed_hal-common_tickers                                                | lp ticker reschedule test                                                                         | 1      | 0      | OK      | 0.23               |
| NUCLEO_L152RE-GCC_ARM | NUCLEO_L152RE | tests-mbed_hal-common_tickers                                                | lp ticker speed test                                                                              | 1      | 0      | OK      | 0.07               |

I try to compare quickly MOTE and NUCLEO code:
- system clock configuration looks different: MOTE is 24 MHz, NUCLEO is 32 MHz ?

bentcooke, 2018-07-27:
@jeromecoutant The lower clock speed appears to be the issue.  The "Test ticker_set_interrupt function" test within the speed test for the lp_ticker is the failure.  Specifically, while + `intf->set_interrupt(0)` takes longer to execute that what is allotted with `MAX_FUNC_EXEC_TIME_US`. 

If I change over to the 32MHz HSI, the execution time scales accordingly and the test passes.

I will make a new PR to increase MAX_FUNC_EXEC_TIME_US to accommodate for the lower clock.

bentcooke, 2018-08-22:
Fixed with #7534.

closed 2018-08-22 by bentcooke