ARMmbed/mbed-os #7560: mbed-cli include issue

AGlass0fMilk, 2018-07-19:
### Description
Hi, I just upgraded to the most recent commit to master. When attempting to build, I am now getting errors saying #includes cannot be found.

I have custom build tools that work on top of mbed-cli. The way I accomplish my builds is by using the --source flag to specify additional include directories and libraries common to all my applications.

A command will look like this for example:

```
['mbed', 'compile', '--build=BUILD/debug', '--source=./', u'--source=/mbed-os', u'--source=/mbed-dsp', u'--source=/hud-core', '--source=../include', '--source=../common', u'--profile=/mbed-os/tools/profiles/debug.json']
```
This used to work fine (all the --source directories would be included in the build. But now the compiler isn't finding header files that reside in these directories...

Has anything changed recently with the build tool that would affect this?

### Issue request type

[ ] Question  
[ ] Enhancement  
[ X ] Bug

0xc0170, 2018-07-20:
My first suspect would be https://github.com/ARMmbed/mbed-os/pull/7183 

@ARMmbed/mbed-os-tools Please review

cmonr, 2018-07-20:
@AGlass0fMilk Would you mind posting the output that you get?

Also, why are some of those strings prefixed with `u`, whereas others are not?

AGlass0fMilk, 2018-07-20:
Well, the custom scripts are written in python and I use Popen to execute `mbed compile`. Popen recommends an input array so I form an array of command options. Printing out the array for verbose mode I guess python appends a `u` to indicate it's an array of `unsigned char`s or whatever.

theotherjimmy, 2018-07-20:
@AGlass0fMilk We recently (in #7183) changed the behavior of include paths from "everything is an include path" to "paths to headers are include paths". You may be seeing that.

theotherjimmy, 2018-07-20:
FYI, it saved ~30% of our include path file length when switching styles.

AGlass0fMilk, 2018-07-20:
The problem I'm seeing is:
I have an 'include' directory for my product that has header files common to all the various programs (bootloader, main application, etc). The implementation for these headers are in a separate 'common' directory. These directories are alongside all the other program directories, similar to the following:

```
product
---> include
------> ble
------> ...
---> common
------> ble
------> ...
---> bootloader
---> main_application
---> factory_reset
.....
```
So I use the `--source` flag to include those common directories when building. Helps me keep my code a bit more organized. What do you recommend for this? The build script can't even find header files in the root of the 'include' directory.

AGlass0fMilk, 2018-07-20:
FYI - This issue is related to something introduced after the recent mbed-os-5.9.3 release. I reverted to that release version and I can still build.

theotherjimmy, 2018-07-20:
Let me see if I can reproduce this on master. BRB

theotherjimmy, 2018-07-20:
@AGlass0fMilk For reproduction value, created the following structure:
```
..
├── blinky
│  ├── <contents of mbed-os-example-blinky
└── temp
   └── header.h

```
and modified main.cpp like so:
```diff
diff --git a/main.cpp b/main.cpp
index 2df9a05..73de4da 100644
--- a/main.cpp
+++ b/main.cpp
@@ -1,4 +1,5 @@
 #include "mbed.h"
+#include "header.h"
 
 DigitalOut led1(LED1);
 
```
ran the following command:
```bash
mbed2 compile --source=./ --source=../temp
```
(`mbed2` is a bash alias for `python2 -m mbed`)
and it did not complain.

Does this simple reproducer cause the same error?

AGlass0fMilk, 2018-07-20:
My build script actually gives absolute paths. Not sure if that would cause the issue to appear, would you be able to test with that?

theotherjimmy, 2018-07-20:
@AGlass0fMilk Sure, no problem.

theotherjimmy, 2018-07-20:
@AGlass0fMilk Reproduced with absolute paths.

theotherjimmy, 2018-07-20:
I very recently (last 2 min) touched the piece of code dealing with those paths, and it fixes it! the last commit or #7559 corrects the absolute path behavior for me.

AGlass0fMilk, 2018-07-20:
Alright, I'll try the PR when it gets merged into master. I'm a bit busy with other development at the moment.

Thanks for looking into this immediately!

theotherjimmy, 2018-09-24:
I think this is resolved.

closed 2018-09-24 by theotherjimmy