ARMmbed/mbed-os #7563: Bootloader header implementation using section

jorisa, 2018-07-20:
### Description
Would it be an idea to instead of using postprocessing in python the application header is included at compile time by using a section on a fixed location (I assume right after the `isr_vector` section would be the most straightforward)

Than you could just do

<...>.ld
```
[...]
SECTIONS
{
    .text :
    {
        KEEP(*(.isr_vector))
        KEEP(*(.header))
[...]
```

header_format.h
```c
struct
{
    uint32_t crc;
    uint64_t sha1;
    uint64_t git_hash;
    uint8_t mbed_major;
    uint8_t mbed_minor;
    uint32_t hw_revision;
    uint32_t fw_version;
    char[32] fw_name;
} header_t;
```

main.cpp
```c
#include "header_format.h"

const header_t __attribute__((section (".header"))) header = {
    .hw_revision= 1,
    .git_hash = GIT_HASH,
    .mbed_major = MBED_MAJOR_VERSION
    .mbed_minor = MBED_MINOR_VERSION
    .fw_version= FIRMWARE_VERSION
    .fw_name= "My supercool application"
};
```

I think this significantly simplifies specifying the header format, it allows both the bootloader and application to access the data stored in the header (without manually having to keep a header and json file in sync), and it makes it easier to also include a header in the bootloader (which can be useful for making sure the firmware type matches the board variant)

### Issue request type
[X] Question  
[ ] Enhancement  
[ ] Bug

theotherjimmy, 2018-07-20:
@jorisa Including the header in the binary of the application complicates linker setup, and still requires python post-build processing to stick the sha1 of the application in the header. This step is much more difficult too.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-291*

closed 2019-10-31 by ciarmcom