ARMmbed/mbed-os #7568: BLE Persistent Bonding w/ iOS Bugs

AGlass0fMilk, 2018-07-20:
### Description
I am on mbed-os-5.9.3 currently.

I'm working on getting persistent bonding to work. Eventually my application will be bonding with smart phones (iOS, Android). I haven't been successful in getting an iPhone to reconnect and pair automatically after a power cycle.

I have a branch on my fork of the ARMmbed/mbed-os-example-ble repository that attempts to demonstrate persistent bonding. I took the unmerged PR for persistent bonding (found here: ARMmbed/mbed-os-example-ble#152) and added the sd-driver also available from ARMmbed.

The filesystem appears to mount correctly but persistent bonding is not functioning as expected with iOS (haven't tested with android). The output I am getting is:

```
Persistent bonding example begin

 PERIPHERAL 

Device address: e0:4d:12:72:95:41
Pairing requested. Authorising.
Link ENCRYPTED
Pairing successful
Disconnected - starting advertising

(reset with JLink)

Persistent bonding example begin

 PERIPHERAL 

Device address: e0:4d:12:72:95:41
Disconnected - starting advertising
Disconnected - starting advertising
Disconnected - starting advertising
Disconnected - starting advertising
Disconnected - starting advertising
Disconnected - starting advertising
Disconnected - starting advertising
Disconnected - starting advertising
```

I attempt to connect to the device from my iPhone after power cycling and an infinite loop of disconnection events starts happening...

Any ideas? You can find my branch here:
https://github.com/AGlass0fMilk/mbed-os-example-ble/tree/bonding-sd-card-example

I have an SD card breakout connected in the following way:
VCC -> 3.3V
GND -> GND
MOSI -> p28
MISO -> p3
SCLK -> p4
CS -> p29

I'm not convinced the persistent bonding feature is actually working...

### Issue request type

[ ] Question  
[ ] Enhancement  
[ X ] Bug

AGlass0fMilk, 2018-07-23:
@paul-szczepanek-arm Have you had any success with persistent BLE bonding? It would be useful to be able to enable more verbose output from the BLE stack (not sure if there's a macro for that).

marcuschangarm, 2018-07-23:
Sounds like the same issue as this one: https://github.com/ARMmbed/mbed-os/issues/7300

AGlass0fMilk, 2018-07-23:
I don't think it's the same issue. I'm trying to use external flash (as currently supported by the security manager) in the form of an SD card, using the ARMmbed/sd-driver driver with littlefs. I'm still not seeing expected functionality (being able to reconnect with encryption automatically).

paul-szczepanek-arm, 2018-07-24:
What status do you get when you call:
preserveBondingStateOnReset()
?

AGlass0fMilk, 2018-07-27:
@pan @paul-szczepanek-arm I tried the pull request in my repo and still the same results - continually disconnecting upon reset and attempting to reconnect on iPhone (also tested with nRF Connect w/ nRF51 dongle).

Another thing I noticed is that the example hard faults if built with the debug.json build profile.

paul-szczepanek-arm, 2018-07-30:
I can build and run the debug profile fine (I tried arm and gcc).
I did try on an iphone and the pairing fails. This isn't a fault with the example, more likely the stack. Looking into this.

paul-szczepanek-arm, 2018-07-31:
Please remove old bonding information from your iphone. That should stop the disconnected loop. We think this is caused by an invalid key being used over and over.

Also pull the recent changes as there was a problem with the example as well where it would not send keys as initiator and not disconnect on the second iteration of the demo.

There is still an issue with iOS we are trying to debug.

AGlass0fMilk, 2018-07-31:
@paul-szczepanek-arm Can you share any details about the issue with iOS that still exists? The demo now successfully re-pairs with my iPhone.

Concerning the disconnect loop:
I have been deleting the bonding information from my iPhone each time I connect (since it would enter the disconnect loop, probably due to invalid keys as you said). I'm not sure what change caused the rebonding to start working.

Which side was using an invalid key over and over?

Thanks for looking into this.

paul-szczepanek-arm, 2018-08-01:
If you comment out the central section in the main() this will work as expected:
connect, pair, encrypt, disconnect, connect, *no pairing*, encrypt (keys already stored)
With the central section the keys exchanged when the device is central don't appear to work when roles are reversed (this might not be a bug, since you can't force the other device to store your key). Will do a sniffer capture to see what is happening and possibly change the example role order to make it work reliably.

paul-szczepanek-arm, 2018-08-02:
There are a couple problems making this issue hard to resolve but the root cause is that phones that use privacy do not get correctly resolved during advertisement by the link controller on NRF52. This creates inconsistency between the device on what keys are valid. This is why everything looks fine as long as the board is just a peripheral and does the advertising and fails when it's central and tries to match against a resolvable address it fails to resolve.

paul-szczepanek-arm, 2018-08-03:
This should work now with the role reversal hint. If both keys (master and slave) aren't sent at the same time phones may create two bonds which might cause issues when roles are reversed since each bond only holds one key rather than there being a single bond with both keys. Closing the issue.

closed 2018-08-03 by paul-szczepanek-arm