ARMmbed/mbed-os #7570: TMPM46B device has duplicate target_id

studavekar, 2018-07-20:
### Description
https://os.mbed.com/platforms/TMPM46B/ devices enumerate with same target ID. We can't have multiple boards connected to same machine. Blocks addition of boards to mbed-os CI.

**mbedls**
output of mbedls when 2 devices are connected to host machine
```
+---------------+----------------------+----------------------------------------------------+--------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point                                        | serial_port  | target_id                                        | daplink_version |
+---------------+----------------------+----------------------------------------------------+--------------+--------------------------------------------------+-----------------+
| TMPM46B       | TMPM46B[0]           | /mnt/pci-0000_00_14_0-usb-0_5_2_3_1_0-scsi-0_0_0_0 | /dev/ttyACM2 | 701300000000000000000000000000000000000097969904 | 0243            |
+---------------+----------------------+----------------------------------------------------+--------------+--------------------------------------------------+-----------------+
```
lsusb -v | grep -i serial 
```
  iSerial                 3 701300000000000000000000000000000000000097969904
  iSerial                 3 701300000000000000000000000000000000000097969904
```
---------------------------------------------------------------
**Target**
TMPM46B  

**DAPLink version:**
```
DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 701300000000000000000000000000000000000097969904
HIC ID: 97969904
Auto Reset: 0
Automation allowed: 0
Overflow detection: 0
Daplink Mode: Interface
Interface Version: 0243
Bootloader Version: 0243
Git SHA: 2f498feb503c4396822ef00f6fbf8c5289dfa9c0
Local Mods: 1
USB Interfaces: MSD, CDC, HID
Bootloader CRC: 0xc06037cb
Interface CRC: 0x07503e06
Remount count: 0
```

**Expected behavior**
It should have unique target ID

**Actual behavior**
It has duplicate target ID

**Steps to reproduce**

1) Connect 2 TMPM46B devices to host
2) use lsusb to check the serial id of device connected. It reports same serial id

**Possible fix**
Needs to implement 

https://github.com/mbedmicro/DAPLink/
void read_unique_id(uint32_t *id);
void create_unique_id(void);

[ ] Question  
[ ] Enhancement  
[X ] Bug  

@maclobdell @ganesh-ramachandran

soramame21, 2018-09-05:
@studavekar @maclobdell I had verified and uploaded the firmware on Jira. please verify.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-5925*

soramame21, 2019-03-11:
@studavekar Could you close this issue please?

studavekar, 2019-03-11:
closing it.

closed 2019-03-11 by studavekar