ARMmbed/mbed-os #7571: mbed_retarget and IAR

Alex-EEE, 2018-07-20:
### Description
Does anyone know the reason / history behind this bit of code inside mbed_retarget, with regards to IAR?

int mbed_getc(std::FILE *_file){
#if defined(__IAR_SYSTEMS_ICC__ ) && (__VER__ < 8000000)
    /*This is only valid for unbuffered streams*/
    int res = std::fgetc(_file);
    if (res>=0){
        _file->_Mode = (unsigned short)(_file->_Mode & ~ 0x1000);/* Unset read mode */
        _file->_Rend = _file->_Wend;
        _file->_Next = _file->_Wend;
    }
    return res;
#else
    return std::fgetc(_file);
#endif
}

I ask b/c this actually breaks getc in version 8.x.  Was there an expectation this would be fixed in version 8.x?  What is this code trying to work around for IAR's sake?

### Issue request type

[ x ] Question  
[ ] Enhancement  
[ ] Bug

0xc0170, 2018-07-23:
From the code, looks like version below 8 has a different implementation . The original one was done some  time ago https://github.com/ARMmbed/mbed-os/commit/bd6836396b8212a28348fdfb4afc14d95d420ca3

There is a difference for IAR in streams . So the above code was working for IAR, but since update 8.x, this broke and has not yet been updated as we do not yet support 8.x

@deepikabhavnani Do you remember this breakage if there is a fix for it?

cc @TTornblom Please review

deepikabhavnani, 2018-07-23:
As mentioned by @0xc0170 - Fix for IAR was done long ago. 

With IAR8 above fix was not working, as `_file->_Wend` `_file->_Rend` were all internal member functions and were not available in header file. 

Some analysis and testing showed that fix was actually not needed for IAR 8.x and so this commit https://github.com/ARMmbed/mbed-os/commit/bcb8779362d9c3d098aad78208b5aab48094efbb#diff-138df4d2b736b2f09dcf330c26bcf9a3 was added.

deepikabhavnani, 2018-07-23:
> I ask b/c this actually breaks getc in version 8.x.

Please share more details on this break

Alex-EEE, 2018-07-23:
Ok, so dug a little deeper.  Looks like after you call getc on a Serial object, putc stops working for good:

Serial pc(USBTX, USBRX); // tx, rx

int main() {
pc.printf("Hello World!\n");  // this prints
    pc.putc('h'); // this prints
    pc.getc(); // this gets the right character
    pc.putc('\n');  // fails, returns -1
    pc.putc('\n');  // fails, returns -1

Alex-EEE, 2018-07-23:
Alternatively, does mbed os have full FILE and C file stream handling built in?  Is there a way to compile with mbed-os's version instead of using IAR's?

0xc0170, 2018-07-24:
> Alternatively, does mbed os have full FILE and C file stream handling built in? Is there a way to compile with mbed-os's version instead of using IAR's?

We do not provide own standard lib, but rely on toolchain one. In this case, what IAR provides. We retarget some functionality only.

Alex-EEE, 2018-07-24:
Makes sense.  Well, it looks like 8.x needs a fix despite the analysis, However I do agree that IAR moved their FILE struct members into a non visible area.  If you guys have any background on what the fix was about for 7.x, and why, I'd be happy to make the fix for 8.x and contribute a PR into your 8.x branch of mbed.

Alex-EEE, 2018-07-24:
Do you have to file a PR for non master branch of mbed-os, or can you directly contribute to feature branches?

deepikabhavnani, 2018-07-25:
Found the PR with issue fix and description: https://github.com/ARMmbed/mbed-os/pull/770

> Do you have to file a PR for non master branch of mbed-os, or can you directly contribute to feature branches?

Solution to this issue can go to master branch

Alex-EEE, 2018-07-25:
Thanks so much guys!!!  I'll try to submit a PR when I fix it.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-290*

closed 2018-10-12 by cmonr