ARMmbed/mbed-os #7572: requirement : read/ write separate lock Serial

jeckypei, 2018-07-22:
### Description

if we Serial as a CLI console .  one CLI thread get the read & write lock , if another thread want to write some to the console, it will be backed by CLI thread, if  we use Serial .  
So we need serial-read lock use different lock with serial-write lock .

I Write such Serial, it can work well  : https://github.com/jeckypei/mbedosBase/blob/master/develop/driver/serial/BetterSerial.cpp

### Issue request type

[ ] Question  
[ ] Enhancement  
[ ] Bug

jeckypei, 2018-07-24:
any update ?    The build-in Serial , use only one lock for serial read & write direction .   it seems not very good sometime .

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-283*

aashishc1988, 2018-10-04:
@jeckypei Can you provide more information on this issue? I understand the problem but I need a way to reproduce it, so an example would be great, :) Just so that we are on the same page.

aashishc1988, 2018-10-17:
@jeckypei , ping!

kjbracey-arm, 2018-10-18:
`Serial` has many problems - use `UARTSerial` instead. If it's for the console, preferably by turning on `platform.stdio-buffered-serial` and just using `printf`/`scanf` etc.

I wasn't really aware of this particular problem.

`UARTSerial` also has a (possibly unnecessary?) single mutex, but it isn't actually held while blocking for input or output, so any read/write interference would be imperceptible.

I don't believe the `BetterSerial` implementation here is portable - the mutexes in the C++ classes are to serialise calls to the HAL - there's no current expectation that a HAL implementation would cope if you let calls to read and write on the same serial object occur overlapping from different threads. I expect it would work on many, but not all. I can imagine some of the trickier platforms like NRF52 not coping.

aashishc1988, 2018-11-14:
@jeckypei Does Kevin's comment answer your question?

jeckypei, 2018-11-15:
OK。 let me try UARTSerial which don't block between input and output .

Thanks .