ARMmbed/mbed-os #7573: requirement : a registerable  CLI

jeckypei, 2018-07-22:
### Description

I have write simple modules  base on tinyShell : 
https://github.com/jeckypei/mbedosBase/wiki/CLI--API
https://github.com/jeckypei/mbedosBase/tree/master/develop/cmd

### Issue request type

[ ] Question  
[ ] Enhancement  
[ ] Bug

0xc0170, 2018-07-23:
I reviewed other issues similar to this one (named as requirement). Can you provide more details (the current implementation, use cases, documentation). Why do you think this is requirement, what this requirement should contain?

jupe, 2018-07-23:
Looks similar with this one: https://github.com/ARMmbed/mbed-client-cli

jupe, 2018-07-24:
It seems that all your requirements have been met - except build-in - which will probably change in future.. Did you try mbed-client-cli ?

jeckypei, 2018-07-24:
i did not try mbed-client-cli .

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-284*

SenRamakri, 2018-12-04:
Hello @jeckypei - Thanks for the suggestion. Unfortunately we don't have any immediate plans to add mbed-cli as build-in feature by default. If you need such functionality please use mbed-client-cli and feel free to expand that by creating PRs to https://github.com/ARMmbed/mbed-client-cli repo.

closed 2018-12-04 by SenRamakri