ARMmbed/mbed-os #7574: requirement :  a good Log API module

jeckypei, 2018-07-22:
### Description

I write  a simple API:  
   https://github.com/jeckypei/mbedosBase/wiki/Log-API

### Issue request type

[ ] Question  
[ ] Enhancement  
[ ] Bug

0xc0170, 2018-07-23:
Please qualify your issue request type. Is this a question or enhancement ?

Have you looked at mbed trace ? See https://github.com/ARMmbed/mbed-os/tree/master/features/frameworks/mbed-trace

SenRamakri, 2018-07-23:
Hi @jeckypei - Thanks much for your feedbacks. Actually we already have a logging API, please take a look at - https://github.com/ARMmbed/mbed-os/blob/master/features/frameworks/mbed-trace/mbed-trace/mbed_trace.h.

Since we already have this implemented please take a look and let us know if it doesnt meet your requirements. For now, I'll close this issue as no more action is required on this.

closed 2018-07-23 by SenRamakri