ARMmbed/mbed-os #7576: STM32L4: ADC read incorrect result on GCC 7.2.0

kevinong0108, 2018-07-22:
### Description

STM32L4 ADC can not read correct value when compiler using GCC 7.2.0 r3, but it works fine on GCC 6.2.0 r4.

Here also confirm with NRF52_DK, it works fine with both GCC toolchain.

#### Verified targets
 - DISCO_L476VG
 - NUCLEO_L476RG
 - NRF52_DK

### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug

0xc0170, 2018-07-23:
@kevinong0108 thanks for reporting. We do not currently support GCC 7.x. However this indicates that there might be a bug in the implementation for L4.

@ARMmbed/team-st-mcd Please review

bcostm, 2018-07-23:
I can't reproduce. I have GCC 7.2.1 20170904 and tested with ci-test AnalogIn + NUCLEO_L476RG:
```
+-----------------------+---------------+--------------------+-----------------------+--------+--------+--------+--------------------+
| target                | platform_name | test suite         | test case             | passed | failed | result | elapsed_time (sec) |
+-----------------------+---------------+--------------------+-----------------------+--------+--------+--------+--------------------+
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-analogin | Analog Input on AIN_0 | 1      | 0      | OK     | 0.07               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-analogin | Analog Input on AIN_1 | 1      | 0      | OK     | 0.04               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-analogin | Analog Input on AIN_2 | 1      | 0      | OK     | 0.05               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-analogin | Analog Input on AIN_3 | 1      | 0      | OK     | 0.04               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-analogin | Analog Input on AIN_4 | 1      | 0      | OK     | 0.05               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-analogin | Analog Input on AIN_5 | 1      | 0      | OK     | 0.05               |
+-----------------------+---------------+--------------------+-----------------------+--------+--------+--------+--------------------+
```

bcostm, 2018-07-24:
I have tested your code and compiled with the Develop profile (it contains the -0s flag) but I still don't see any problem. All analog pins have the correct values.

I have no idea why it is not working for you ?

kevinong0108, 2018-07-24:
#### Environment
- Toolchain
https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads
7-2017-q4-major (GCC 7.2.1 20170904)
- mbed-os 5.9.3
- mbed-cli
mbed compile --profile=./mbed-os/tools/profiles/develop.json -m NUCLEO_L476RG
- Simple Test Code as main.c
A0 connect to GND, printf shows non-zero value (around 160)

Here is my build bin: 
[mbed-os_NUCLEO_L476RG_adc_test.zip](https://github.com/ARMmbed/mbed-os/files/2225768/mbed-os_NUCLEO_L476RG_adc_test.zip)

Thanks!!

bcostm, 2018-07-25:
I have tested your bin file and I can see the 160 value on A0 pin too.

I have checked again using mbed-os-5.9.3 (because I was on master the previous time) and using the develop profile, your main.c file but still no issue (I have 0 on the A0 pin).

So, we have the same GCC version, the same mbed-os version, the same main file, it is not the board... 

For the tools I have:
mbed-cli         1.7.5
mbed-greentea    1.4.0
mbed-host-tests  1.4.0

Which versions do you have ? ($ pip list | grep mbed)

kevinong0108, 2018-07-25:
I use mbed-cli 1.7.5 too.
I saw $ and grep, I think I found the different between us. I am use windows environment...

I need some time to create the build environment to verify it.
Thanks your test and reply!

kevinong0108, 2018-07-25:
#### Environment
- Ubuntu 64-bit 16.04.4 Server
- GCC_ARM version 7-2017-q4-majo (GCC 7.2.1)
- mbed os 5.9.3
- $ pip list | grep mbed
mbed-cli         1.7.5
mbed-greentea    1.4.0
mbed-host-tests  1.4.0
mbed-ls          1.5.0

#### Build Flow
- Install Ubuntu 64-bit 16.04.4
- Install mbed-cli
- Install GNU ARM Embedded toolcain 6 and 7
- $ mbed new mbed-os-adc
- create main.c (Simple Test Code)
- [mbed] Local config
GCC_ARM_PATH=/opt/toolchain/gcc-arm-none-eabi-7-2017-q4-major/bin/
TOOLCHAIN=GCC_ARM
TARGET=NUCLEO_L476RG
- $ mbed compile --profile=mbed-os/tools/profiles/develop.json

#### Result
Same as windows version...

But when I change toolchain to "GNU Arm Embedded toolchain (GCC) version 6", everything is OK.
I have no idea why it is not working for me...

bcostm, 2018-07-26:
I am also on Windows but I used Git bash window.

But I can reproduce the issue on my side now. The GCC version written in mbed_settings.py file was pointing to the V6 and not V7...

At least it is reproduced, but now we need to find why...

jeromecoutant, 2018-08-06:
Hi
Maybe we can close this issue ?
Thx

closed 2018-08-06 by kevinong0108