ARMmbed/mbed-os #7588: STM32 Nucleo-L432KC USBDevice library

sstaub, 2018-07-24:
Is it possible to support the Nucleo-L432KC in the USBDevice library ?

sstaub, 2018-07-24:
After a short look inside the ongoing usb development, in the USBPhy_STM32.cpp file,
the Nucleo-L432KC is missing, also the NUCLEO-L433RC-P board.

bcostm, 2018-07-25:
The plan is to finalize first the new USB development on a couple of STM32 board (NUCLEO_F207ZG, ...). Then the deployment on all STM32 boards supporting USB will be done but I no delivery date for the moment. @sstaub, community help is welcome for this task. So, feel free to propose PRs for these boards if you want.

0xc0170, 2018-07-25:
@sstaub Is your question answered?

sstaub, 2018-07-25:
Yes, I will wait. Maybe I will find some time to make a PR, but I'm not firm with the HAL.

closed 2018-10-04 by adbridge

sstaub, 2018-10-04:
Why is this issue closed? What have it to do with the closed_in_jira?

adbridge, 2018-10-04:
@sstaub You asked a question, @0xc0170 asked you if it had been answered and you said 'Yes' so the issue was closed....

sstaub, 2018-10-04:
Misunderstood, there should a support for the Nucleo-L432KC, it is an issue for further USB development.

kubaraczkowski, 2019-10-04:
Hi, accidentally exactly a year after the last message - is there a plan to add USB support to the L432KC device ?
The new USB stack has been released a few months ago, working great on L476...

jeromecoutant, 2019-10-04:
Hi
I have started dev here: https://github.com/jeromecoutant/mbed/tree/DEV_USB_STM32
But it's not working very well yet... Any help is welcome!

mimok, 2020-02-24:
Hi Jerome,

Do you plan to merge your on-going work on STM USB drivers in the Mbed official repo? 

I'm developing a Murata Type ABZ based board which is using USB port as main communication interface. The lack of USB support in mbed is a real pain. Until now, I'm using UART as a workaround which is not the intended use case.