ARMmbed/mbed-os #7589: How to increase the RAM size for a target in the linker script ?

LaurentLouf, 2018-07-24:
### Description

Working with a STM32F103xC, I'm actually using the closest target that is available in the most recent version of mbed-os, the target NUCLEO_F103RB. The main difference between the two is the memory, the xC having more flash and RAM than the xB series. So, in the first time, I would like to create a custom target, but I'm a bit stuck when I look at the linker script : 
```
MEMORY
{ 
  FLASH (rx) : ORIGIN = 0x08000000, LENGTH = 128K
  RAM (rwx) : ORIGIN = 0x200000EC, LENGTH = 20K - 0xEC
}

```

Changing the flash seems pretty straightforward, I just have to put 256K instead of 128K. But for the RAM, I'm a bit lost. I could just change the size, but there seems to be some kind of offset, and I don't have any idea how it is defined or where. So the question is : how to increase the RAM to match my board, a STM32F103RCT6 to be precise ? And the enhancement would be to integrate this target directly in the repository, to avoid having to duplicate a similar target with some adjustments.

### Issue request type

[X] Question  
[X] Enhancement  
[ ] Bug

mstfkaratas, 2018-07-25:
According to datasheet of your mcu there is 60+16 interrupt lines, so you should reserve 76*4=304 bytes which is 0x130. LENGTH should be XK - 0x130 where XK is your RAM  size.
I think you also need to edit vector table in startup.S file but I dont know how it is generated, maybe you can look at cubemx files of your mcu.

Mustafa

0xc0170, 2018-07-25:
@ARMmbed/team-st-mcd Please review

0xc0170, 2018-07-25:
As mentioned, need to adjust RAM (leave space for vector table - copied during the startup to RAM). If number of vectors is different, need to edit the vector table as well in the startup file.

bcostm, 2018-07-25:
Yes for the RAM size, please follow @mstfkaratas explanation (thanks).

For the startup, linker files, registers files etc... we use the files from ST CubeF1 package (they are in the CMSIS/Device/ST folders) . But you need to take care to take the same Cube version as the actual one used in mbed and not necessarily the last one. As you can have some compatibility issues. Note also that the startup and IAR linker files are also modified for mbed. The better is to make a diff between F103RB files and those in Cube to see the differences...

For the F1 devices, we use Cube V1.5.0 in mbed. Please find the CMSIS/Device/ST/STM32F1xx folder corresponding to this version:
[STM32F1xx.zip](https://github.com/ARMmbed/mbed-os/files/2227380/STM32F1xx.zip)

LaurentLouf, 2018-07-25:
Thanks everyone and especially @mstfkaratas , I wouldn't have thought of the interrupt lines without you. Thanks for the suggestion @bcostm but although I didn't pay any attention to the version of Cube files, I noticed that the F103xC isn't present in those files. We have 103x6, 103xB, 103xE and 103xG. so the xC and xD versions are missing (high density lines with 256 and 384kB of flash)

bcostm, 2018-07-25:
>  I noticed that the F103xC isn't present in those files
Because these devices were not supported in Cube V1.5.0. It is then required to update the Cube version in mbed with the latest version. But this will require some time as we need to perform a full non-regression testing. In the mean time you can take the files you need in the latest Cube version and check if it is ok on your board.

0xc0170, 2018-07-25:
@LaurentLouf Does that answer all your questions?

bcostm, 2018-07-25:
Note that I'm actually updating the CubeF1. Non-reg tests are in progress...

LaurentLouf, 2018-07-25:
@0xc0170 yeah, except that unfortunately the V1.6.0 of CubeF1 does not seem to have the linker and startup files for the F103xC, so I'm out of luck on that end.
[image]

LaurentLouf, 2018-07-26:
So does that seem right ? It feels like it (and it compiles, and flashes, and runs) but on the RAM side, it seems that it still considers that we have only 20K. I have made a test by changing the size of a static buffer, and it appears that static RAM + reserved heap size is at 19152 bytes, see the logs after : 
> Total Static RAM memory (data + bss): 10388 bytes
> Heap size: reserved 8764, max 3824, current 3824

> Total Static RAM memory (data + bss): 10556 bytes
> Heap size: reserved 8596, max 3824, current 3824

If all went well, it should in theory be close to 48K, not 20K. I'm compiling with mbed CLI 1.7.3 and have recompiled using the `--clean` flag just in case, same result. My `.mbed` file is as follow : 
> 
> TARGET=NUCLEO_F103RB
> TOOLCHAIN=GCC_ARM
> ROOT=.

And I am indeed modifying in place for the moment, so directly in the NUCLEO_F103RB target folder. So I'm a bit short on ideas.

[startup_stm32f103xb.S.txt](https://github.com/ARMmbed/mbed-os/files/2231548/startup_stm32f103xb.S.txt)
[STM32F103XB.ld.txt](https://github.com/ARMmbed/mbed-os/files/2231549/STM32F103XB.ld.txt)

bcostm, 2018-07-26:
Can you try again maybe doing a "rm -rf BUILD" instead of the --clean option ?

LaurentLouf, 2018-07-26:
@bcostm no success unfortunately. Must be something stupid but I don't see it.

bcostm, 2018-07-26:
I am not sure if it will do something but did you update the TARGET_STM/mbed_rtx.h file ?

bcostm, 2018-07-26:
Concerning the stm32f103xc.h file, I didn't find it anywhere in Cube firmware. But looking inside the stm32f1xx.h I saw this at line 91:

`/* #define STM32F103xE */    /*!< STM32F103RC, STM32F103VC, STM32F103ZC, STM32F103RD, STM32F103VD, STM32F103ZD, STM32F103RE, STM32F103VE and STM32F103ZE */`

And at the bottom of this file you can see that there is also no mention of the stm32f103xc.h file.

That means for me that you can use the stm32f103xe.h file for your device.

jeromecoutant, 2018-08-06:
Hi

As a conclusion, check https://os.mbed.com/teams/ST/wiki/steps-to-create-a-new-STM32-platform

And:

- F1 cube driver update : #7606

- linker file:

MEMORY
{ 
  FLASH (rx) : ORIGIN = 0x08000000, LENGTH = 256K
  RAM (rwx) : ORIGIN = 0x200000EC, LENGTH = 64K - 0x130
}

- stm32f1xx.h : uncomment the correct MCU

- TARGET_STM/mbed_rtx.h : add something like

#elif defined(TARGET_STM32F103RC)
#define INITIAL_SP    (0x20010000UL)

- targets.json : in extra_labels_add, replace STM32F103RB by STM32F103RC

0xc0170, 2018-08-07:
@LaurentLouf Does that answer your question?

LaurentLouf, 2018-08-07:
Yes it does, I'll reopen if I have time to do it the clean way and any question.

closed 2018-08-07 by LaurentLouf

hgalindo, 2020-03-26:
> Hi
> 
> As a conclusion, check https://os.mbed.com/teams/ST/wiki/steps-to-create-a-new-STM32-platform
> 
> And:
> 
> * F1 cube driver update : #7606
> * linker file:
> 
> MEMORY
> {
> FLASH (rx) : ORIGIN = 0x08000000, LENGTH = 256K
> RAM (rwx) : ORIGIN = 0x200000EC, LENGTH = 64K - 0x130
> }
> 
> * stm32f1xx.h : uncomment the correct MCU
> * TARGET_STM/mbed_rtx.h : add something like
> 
> #elif defined(TARGET_STM32F103RC)
> #define INITIAL_SP (0x20010000UL)
> 
> * targets.json : add an new target with STM32F103RC in extra_labels_add

Hi, 
I was trying to do follow these steps, but it seems for the mbed-os 5.15.1 should be different.
Also https://os.mbed.com/teams/ST/wiki/steps-to-create-a-new-STM32-platform is not totally compatible with 5.15.1.

Can you please update the procedure?

Thanks!

jeromecoutant, 2020-03-26:
@hgalindo 
yes...  I agree wiki page is not fully up to date...
Note that mbed-os\tools\targets\STM32_gen_PeripheralPins.py script to generate PeripheralPins.c and the end of PinNames.h is maintained.