ARMmbed/mbed-os #7594: Windows makefile builds no longer work

cmonr, 2018-07-25:
### Description

Exporting projects to a `make_*` ide, and running `make` no longer appears to work since https://github.com/ARMmbed/mbed-os/pull/7183 came in (confirmed via `git bisect`).

This has been validated with `make_armc5` and `make_gcc_arm` "IDEs".

PRs https://github.com/ARMmbed/mbed-os/pull/7583 and https://github.com/ARMmbed/mbed-os/pull/7559 do not fix the issue.

### Output
```
C:\Users\redacted\test\mbed-os-example-blinky>(cd mbed-os && git checkout master && cd ..)
Already on 'master'
Your branch is up-to-date with 'origin/master'.

C:\Users\redacted\test\mbed-os-example-blinky>pipenv run mbed export -i make_gcc_arm -m K64F --clean && make
Scan: mbed-os-example-blinky
C:/Users/root/test/mbed-os-example-blinky/makefile:865: warning: overriding commands for target `.s.o'
C:/Users/root/test/mbed-os-example-blinky/makefile:857: warning: ignoring old commands for target `.s.o'
Compile: main.cpp
../main.cpp:1:18: fatal error: mbed.h: No such file or directory
 #include "mbed.h"
                  ^
compilation terminated.
make[1]: *** [main.o] Error 1
make: *** [all] Error 2

C:\Users\redacted\test\mbed-os-example-blinky>pipenv run mbed export -i make_armc5 -m K64F --clean && make
Scan: mbed-os-example-blinky
C:/Users/root/test/mbed-os-example-blinky/makefile:861: warning: overriding commands for target `.s.o'
C:/Users/root/test/mbed-os-example-blinky/makefile:852: warning: ignoring old commands for target `.s.o'
Compile: main.cpp
"../main.cpp", line 1: Error:  #5: cannot open source input file "mbed.h": No such file or directory
../main.cpp: 0 warnings, 1 error
make[1]: *** [main.o] Error 1
make: *** [all] Error 2

C:\Users\redacted\test\mbed-os-example-blinky>(cd mbed-os && git checkout pr7583 && cd ..)
Switched to branch 'pr7583'

C:\Users\redacted\test\mbed-os-example-blinky>pipenv run mbed export -i make_gcc_arm -m K64F --clean && make
Scan: mbed-os-example-blinky
C:/Users/root/test/mbed-os-example-blinky/makefile:865: warning: overriding commands for target `.s.o'
C:/Users/root/test/mbed-os-example-blinky/makefile:857: warning: ignoring old commands for target `.s.o'
Compile: main.cpp
../main.cpp:1:18: fatal error: mbed.h: No such file or directory
 #include "mbed.h"
                  ^
compilation terminated.
make[1]: *** [main.o] Error 1
make: *** [all] Error 2

C:\Users\redacted\test\mbed-os-example-blinky>pipenv run mbed export -i make_armc5 -m K64F --clean && make
Scan: mbed-os-example-blinky
C:/Users/root/test/mbed-os-example-blinky/makefile:861: warning: overriding commands for target `.s.o'
C:/Users/root/test/mbed-os-example-blinky/makefile:852: warning: ignoring old commands for target `.s.o'
Compile: main.cpp
"../main.cpp", line 1: Error:  #5: cannot open source input file "mbed.h": No such file or directory
../main.cpp: 0 warnings, 1 error
make[1]: *** [main.o] Error 1
make: *** [all] Error 2

C:\Users\redacted\test\mbed-os-example-blinky>(cd mbed-os && git checkout pr7559 && cd ..)
Switched to branch 'pr7559'

C:\Users\redacted\test\mbed-os-example-blinky>pipenv run mbed export -i make_gcc_arm -m K64F --clean && make
Scan: mbed-os-example-blinky
C:/Users/root/test/mbed-os-example-blinky/makefile:1305: warning: overriding commands for target `.s.o'
C:/Users/root/test/mbed-os-example-blinky/makefile:1297: warning: ignoring old commands for target `.s.o'
Compile: main.cpp
arm-none-eabi-g++.exe: fatal error: no input files
compilation terminated.
make[1]: *** [main.o] Error 1
make: *** [all] Error 2

C:\Users\redacted\test\mbed-os-example-blinky>pipenv run mbed export -i make_armc5 -m K64F --clean && make
Scan: mbed-os-example-blinky
C:/Users/root/test/mbed-os-example-blinky/makefile:1097: warning: overriding commands for target `.s.o'
C:/Users/root/test/mbed-os-example-blinky/makefile:1088: warning: ignoring old commands for target `.s.o'
Compile: main.cpp
Error: C4079E: armcc command with no effect

make[1]: *** [main.o] Error 1
make: *** [all] Error 2

C:\Users\redacted\test\mbed-os-example-blinky>
```

### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug

theotherjimmy, 2018-07-25:
Yeah that's fixed by these commits: https://github.com/ARMmbed/mbed-os/pull/7559/commits/881803ca9b09afa72249aa9881a8468fe0b69d46 and https://github.com/ARMmbed/mbed-os/pull/7559/commits/780352b05dcb87a60b8d91bb33839f7b9175c343

theotherjimmy, 2018-07-26:
@0xc0170 That looks correct. I would like you to test the 544c2426c merge commit as well, if you get the chance.

0xc0170, 2018-07-26:
on that commit `544c2426c `

```
*   1cd80b0e7 - (HEAD -> master, upstream/master, theotherjimmy-make-cmd-files) Merge pull request #128 from adbridge/master (3 weeks ago) <Martin Kojtal>
```
```
mbed-os-example-blinky>make
mbed-os-example-blinky/makefile:1323: warning: overriding commands for target `.s.o'
mbed-os-example-blinky/makefile:1315: warning: ignoring old commands for target `.s.o'
Compile: main.cpp
arm-none-eabi-g++.exe: fatal error: no input files
compilation terminated.
make[1]: *** [main.o] Error 1
make: *** [all] Error 2
```

theotherjimmy, 2018-07-26:
Yeah. I'm testing with the exceptionally slow windows cmd now. That's a quoting issue. The commits I mentioned do fix the problem, so I rebased them onto the other PR. That PR works, mostly.

theotherjimmy, 2018-07-26:
see #7608

cmonr, 2018-07-26:
Closing since the original regression is now fixed.

Other issues (https://github.com/ARMmbed/mbed-os/issues/6335, https://github.com/ARMmbed/mbed-os/issues/7129) remain open.

closed 2018-07-26 by cmonr