ARMmbed/mbed-os #7607: STM32F4 & F1 add hardware CRC support

pilotak, 2018-07-26:
### Description
Since STM32F4 and F1 have CRC engine it would be nice to support it mbedOS. I can see that there is already support for L0,L4,F0,F3,F7 it would be nice to have them all covered

### Issue request type

[ ] Question  
[X] Enhancement  
[ ] Bug

bcostm, 2018-07-27:
For information, this is the list of ST boards having the `CRC ` feature defined in targets.json as of today (in green):
[image]

CRC has been added in PR #6708 but I don't know why it has not been enabled for all STM32 platforms in targets.json ?

pilotak, 2018-07-27:
It's a shame that F4 is not enabled, i tried to enable it manually on F429ZI but it doesn't compile

bcostm, 2018-07-30:
I confirm the build error with the NUCLEO_F429ZI:
```
[Error] mbed_crc_api.c@41,18: 'CRC_HandleTypeDef {aka struct <anonymous>}' has no member named 'InputDataFormat'
[Error] mbed_crc_api.c@41,50: 'CRC_INPUTDATA_FORMAT_BYTES' undeclared (first use in this function); did you mean 'IS_I2S_DATA_FORMAT'?
[Error] mbed_crc_api.c@42,18: 'CRC_HandleTypeDef {aka struct <anonymous>}' has no member named 'Init'
[Error] mbed_crc_api.c@42,50: 'DEFAULT_POLYNOMIAL_ENABLE' undeclared (first use in this function)
[Error] mbed_crc_api.c@43,18: 'CRC_HandleTypeDef {aka struct <anonymous>}' has no member named 'Init'
[Error] mbed_crc_api.c@43,50: 'DEFAULT_INIT_VALUE_DISABLE' undeclared (first use in this function); did you mean '__HAL_TIM_MOE_DISABLE'?
[Error] mbed_crc_api.c@44,18: 'CRC_HandleTypeDef {aka struct <anonymous>}' has no member named 'Init'
[Error] mbed_crc_api.c@45,18: 'CRC_HandleTypeDef {aka struct <anonymous>}' has no member named 'Init'
[Error] mbed_crc_api.c@45,50: 'CRC_POLYLENGTH_32B' undeclared (first use in this function); did you mean 'IRDA_WORDLENGTH_9B'?
[Error] mbed_crc_api.c@46,18: 'CRC_HandleTypeDef {aka struct <anonymous>}' has no member named 'Init'
[Error] mbed_crc_api.c@47,30: 'CRC_INPUTDATA_INVERSION_BYTE' undeclared (first use in this function); did you mean 'CRC_INPUTDATA_FORMAT_BYTES'?
[Error] mbed_crc_api.c@48,11: 'CRC_INPUTDATA_INVERSION_NONE' undeclared (first use in this function); did you mean 'CRC_INPUTDATA_INVERSION_BYTE'?
[Error] mbed_crc_api.c@49,18: 'CRC_HandleTypeDef {aka struct <anonymous>}' has no member named 'Init'
[Error] mbed_crc_api.c@50,31: 'CRC_OUTPUTDATA_INVERSION_ENABLE' undeclared (first use in this function); did you mean 'CRC_OUTPUTDATA_INVERSION_ENABLED'?
[Error] mbed_crc_api.c@51,11: 'CRC_OUTPUTDATA_INVERSION_DISABLE' undeclared (first use in this function); did you mean 'CRC_OUTPUTDATA_INVERSION_DISABLED'?
```
I will have a look.

bcostm, 2018-07-30:
I could pass the build by simply commenting the initialization of the CRC init structure but the mbed-os CRC tests are FAIL... It is unfortunately not enough and more work is required to use correctly the CRC peripheral on the F4 devices (and surely it will be the same for F1 and F2).

jeromecoutant, 2018-08-10:
Hi

Yes, CRC driver version for STM32F1/F2/F4 and L1 is older than other STM32 family.
We need to update the current MBED implementation to match all family specific CRC feature.

For ex, in https://github.com/ARMmbed/mbed-os/blob/54f40a0f4ffa9888346d82b934d6d783d37048a7/targets/TARGET_STM/mbed_crc_api.c#L56

U8 input data are cast to U32 to match HAL_CRC_Accumulate API, and InputDataFormat have been previously set to Bytes
With F4, there is no possibility to set byte input format, it has to be Word, so this hal_crc_compute_partial has to be updated.

Note that this is not currently planned in the ST short term activities,
so help are welcomed

screamerbg, 2018-09-10:
@jeromecoutant Do you have an idea about the timeline for this?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-5945*

closed 2019-10-31 by ciarmcom