ARMmbed/mbed-os #7610: AnalogIn fails on I.MXRT1050

maclobdell, 2018-07-26:
### Description

I'm measuring an analog waveform from 0 to 3.2V using the following code.

void AnalogInput_Test()
{
    AnalogIn ain(A0);

    for(int i=0;i<10;i++)
    {
      printf("Ad = %d\n\r", ain.read());
      printf("Af = %f\n\r",ain.read());
      wait_ms(1000);
    }
}

First off, the decimal result is stuck at the same value.  

The float result does seem to roughly follow the actual analog waveform, but the number is so small.  Is there a gain value that needs to be set?  

ain=-2147481196
ain=0.006363
ain=-2147481196
ain=0.006485
ain=-2147481196
ain=0.017304
ain=-2147481196
ain=0.018311
ain=-2147481196
ain=0.020356
ain=-2147481196
ain=0.040040
ain=-2147481196
ain=0.040131
ain=-2147481196
ain=0.042481
ain=-2147481196
ain=0.059663
ain=-2147481196
ain=0.059739

[ ] Question  
[ ] Enhancement  
[X] Bug  

@ARMmbed/team-nxp

mmahadevan108, 2018-08-02:
This has been fixed via PR: https://github.com/ARMmbed/mbed-os/pull/7429
Below is the test result with the current mbed-os master:

PS C:\ci-test-shield.git> mbed test -n tests-api-analogin* --app-config .\mbed_app.json -t ARM -m MIMXRT1050_EVK
Building library mbed-build (MIMXRT1050_EVK, ARM)
Scan: ci-test-shield.git
Building project AnalogIn (MIMXRT1050_EVK, ARM)
Scan: ARM
Scan: AnalogIn
Link: AnalogIn
Elf2Bin: AnalogIn
+------------------+-------+-------+--------+
| Module           | .text | .data |   .bss |
+------------------+-------+-------+--------+
| TESTS\API        |  3596 |     0 |    212 |
| [lib]\c_w.l      | 11593 |    16 |    348 |
| [lib]\cpprt_w.l  |    42 |     0 |      0 |
| [lib]\fz_wv.l    |    18 |     0 |      0 |
| [lib]\m_wv.l     |    48 |     0 |      0 |
| anon$$obj.o      |    32 |     0 |  99328 |
| mbed-os\drivers  |  1419 |     0 |     40 |
| mbed-os\features |  7649 |   140 |    412 |
| mbed-os\hal      |  1724 |     6 |     64 |
| mbed-os\platform |  3399 |   100 |    516 |
| mbed-os\rtos     | 11397 |  2313 |   4592 |
| mbed-os\targets  | 15894 |   166 |    312 |
| Subtotals        | 56811 |  2741 | 105824 |
+------------------+-------+-------+--------+
Total Static RAM memory (data + bss): 108565 bytes
Total Flash memory (text + data): 59552 bytes

Image: BUILD/tests/MIMXRT1050_EVK/ARM/TESTS/API/AnalogIn/AnalogIn.bin

Memory map breakdown for built projects (values in Bytes):
+----------+----------------+-----------+------------+-------------+
| name     | target         | toolchain | static_ram | total_flash |
+----------+----------------+-----------+------------+-------------+
| AnalogIn | MIMXRT1050_EVK | ARM       |     108565 |       59552 |
+----------+----------------+-----------+------------+-------------+

Build successes:
  * MIMXRT1050_EVK::ARM::MBED-BUILD
  * MIMXRT1050_EVK::ARM::TESTS-API-ANALOGIN
mbedgt: greentea test automation tool ver. 1.4.0
mbedgt: test specification file 'C:\ci-test-shield.git\BUILD\tests\MIMXRT1050_EVK\ARM\test_spec.json' (specified with --
test-spec option)
mbedgt: using 'C:\ci-test-shield.git\BUILD\tests\MIMXRT1050_EVK\ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
mbedgt: processing target 'MIMXRT1050_EVK' toolchain 'ARM' compatible platforms... (note: switch set to --parallel 1)
mbedgt: test case filter (specified with -n option)
        tests-api-analogin*
        test filtered in 'tests-api-analogin'
mbedgt: running 1 test for platform 'MIMXRT1050_EVK' and toolchain 'ARM'
mbedgt: mbed-host-test-runner: started
mbedgt: checking for GCOV data...
mbedgt: test on hardware with target id: 0227000041114e450004300ac2040020df91000097969900
mbedgt: test suite 'tests-api-analogin' .............................................................. OK in 15.69 sec
        test case: 'Analog Input on AIN_0' ........................................................... OK in 0.33 sec
        test case: 'Analog Input on AIN_1' ........................................................... OK in 0.32 sec
        test case: 'Analog Input on AIN_2' ........................................................... OK in 0.32 sec
        test case: 'Analog Input on AIN_3' ........................................................... OK in 0.33 sec
        test case: 'Analog Input on AIN_4' ........................................................... OK in 0.33 sec
        test case: 'Analog Input on AIN_5' ........................................................... OK in 0.33 sec
mbedgt: test case summary: 6 passes, 0 failures
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.6107744395
mbedgt: test suite report:
+--------------------+----------------+--------------------+--------+--------------------+-------------+
| target             | platform_name  | test suite         | result | elapsed_time (sec) | copy_method |
+--------------------+----------------+--------------------+--------+--------------------+-------------+
| MIMXRT1050_EVK-ARM | MIMXRT1050_EVK | tests-api-analogin | OK     | 15.69              | default     |
+--------------------+----------------+--------------------+--------+--------------------+-------------+
mbedgt: test suite results: 1 OK
mbedgt: test case report:
+--------------------+----------------+--------------------+-----------------------+--------+--------+--------+---------
-----------+
| target             | platform_name  | test suite         | test case             | passed | failed | result | elapsed_
time (sec) |
+--------------------+----------------+--------------------+-----------------------+--------+--------+--------+---------
-----------+
| MIMXRT1050_EVK-ARM | MIMXRT1050_EVK | tests-api-analogin | Analog Input on AIN_0 | 1      | 0      | OK     | 0.33
           |
| MIMXRT1050_EVK-ARM | MIMXRT1050_EVK | tests-api-analogin | Analog Input on AIN_1 | 1      | 0      | OK     | 0.32
           |
| MIMXRT1050_EVK-ARM | MIMXRT1050_EVK | tests-api-analogin | Analog Input on AIN_2 | 1      | 0      | OK     | 0.32
           |
| MIMXRT1050_EVK-ARM | MIMXRT1050_EVK | tests-api-analogin | Analog Input on AIN_3 | 1      | 0      | OK     | 0.33
           |
| MIMXRT1050_EVK-ARM | MIMXRT1050_EVK | tests-api-analogin | Analog Input on AIN_4 | 1      | 0      | OK     | 0.33
           |
| MIMXRT1050_EVK-ARM | MIMXRT1050_EVK | tests-api-analogin | Analog Input on AIN_5 | 1      | 0      | OK     | 0.33
           |
+--------------------+----------------+--------------------+-----------------------+--------+--------+--------+---------
-----------+
mbedgt: test case results: 6 OK
mbedgt: completed in 20.95 sec
PS C:\ci-test-shield.git>

0xc0170, 2018-08-03:
@maclobdell Please retest. The initial report does not provide what version of Mbed OS was tested.

maclobdell, 2018-09-10:
@mmahadevan108  I just tested Mbed OS 5.9.7 and got failures.  Please confirm that a patch was submitted to fix the AnalogIn tests.

maclobdell, 2018-09-10:
I fixed a connection problem on my CI Test Shield.  It is working now.  Closing this issue.  Thanks!

closed 2018-09-10 by maclobdell