ARMmbed/mbed-os #7611: Support for SD Card on I.MXRT1050_EVK(B)

maclobdell, 2018-07-27:
### Description

The SD Card interface on the I.MXRT1050_EVK(B) board is connected to the following signals for SDHC function.  
            MCU Pin               SDHC
            GPIO_SD_B0_04    SD_DAT2               
            GPIO_SD_B0_05    SD_DAT3            
            GPIO_SD_B0_03    SD_DAT1             
            GPIO_SD_B0_02    SD_DAT0          
            GPIO_SD_B0_01    SD_CLK           
            GPIO_SD_B0_00    SD_CMD       

Mbed OS currently does not have built-in support for SDHC (4 data lines).  It should be possible to use the uSDHC driver from the MCUXpresso SDK directly, but, for ease of use, using regular SPI mode would be nice.  

It looks like the same pins also have regular LPSPI functionality, but the SPI function signals don't line up correctly.

The reference manual says the uSDHC peripheral supports 1-bit SD mode.  

If the uSDHC can be configured for SPI operation, then it would be great to have a SPI implementation for the following pins in this order.  

The pin mapping should be something like this

         GPIO_SD_B0_01   SPI_CLK
         GPIO_SD_B0_05   SPI_CS
         GPIO_SD_B0_02   SPI_MISO
         GPIO_SD_B0_00   SPI_MOSI

Mbed Os 5.9.3 is the latest release at the time of this ticket.

### Issue request type

[ ] Question  
[X ] Enhancement  
[ ] Bug

maclobdell, 2018-07-27:
@ARMmbed/team-nxp

mmahadevan108, 2018-09-27:
uSDHC does not support SPI mode. Are there plans to support non-SPI controllers inside mbed-os SDDriver?

JojoS62, 2018-09-27:
I have started some work on a SDIOBlockdevice, see issue #6384. There is also a link to my fork with some code for a STM32F4 implementation. I've used the STM HAL, but in an additional layer. This could be common to fit also for other targets. The functions are basc read/write/ease and infos. There is a reference to a HAL structure that needs to be removed.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-5949*

closed 2019-10-31 by ciarmcom