ARMmbed/mbed-os #7612: Query: Mem-Size (heap memory) in Lwip

deepikabhavnani, 2018-07-27:
### Description

Heap Size in Lwip stack has huge difference between various platforms, was curious to know why we have such wide range of heap size (16K-50K) for different targets with same features in stack?

### Issue request type
[X] Question  
[ ] Enhancement  
[ ] Bug

0xc0170, 2018-07-27:
@ARMmbed/mbed-os-ipcore

mikaleppanen, 2018-07-31:
On lwip, the heap is used for outgoing messages and socket buffer. Larger sizes mean that lwip can store more application socket data on its own buffers.

Also depending on ethernet driver implementation, the driver may use lwip heap for incoming data e.g. if it needs continuous memory buffers for DMA transfer. So for drivers using continuous buffers, heap is larger than on platforms using chained buffers from buffer pool.

deepikabhavnani, 2018-07-31:
@mikaleppanen - Thanks for clarification, since this is dependent on underlying platform, I believe is useful information for users.
Do we have any documentation describing suggested heap size based on requirement (outgoing only vs incoming + outgoing buffers). Calculations stating how many buffers are used in each direction and buffer sizes.

AnotherButler, 2018-08-10:
@deepikabhavnani I think the closest we have is our documentation on heap stats. I'll create an internal JIRA task for this.

deepikabhavnani, 2018-08-10:
@AnotherButler - Heap stats is not related to heap in use by lwip. @mikaleppanen along with @hasnainvirk and @kjbracey-arm  did work on optimizing this heap value and can share more insights on calculation and optimization.

We should capture those in some documentation.

closed 2018-08-21 by deepikabhavnani