ARMmbed/mbed-os #7613: Export via os.mbed.com does not include mbed_config.h

janjongboom, 2018-07-27:
### Description

1. Go to https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-blinky/export.
1. Use `LPC1768` as target, `uvision5` as IDE.
1. Export.
1. Inspect the ZIP.
1. `mbed_config.h` is missing and thus builds fail.

Mbed CLI does not have this issue. Export via Online Compiler also does not have this issue.

Note that this is our most popular board and our most popular IDE. If we're going to break this export flow or never check it, we should just remove it.

@immunda @theotherjimmy 

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

theotherjimmy, 2018-07-27:
@janjongboom 

> If we're going to break this export flow or never check it, we should just remove it.

Don't be so incendiary. I check that particular export (exactly that one) when I update the tools in the online compiler.

janjongboom, 2018-07-30:
@theotherjimmy Sorry, I was a bit frustrated last week. So I'm wondering where the difference comes from, as importing the project in Online Compiler and then exporting to uVision5 target actually adds mbed_config.h...

theotherjimmy, 2018-07-30:
@janjongboom No harm done. I understand the frustration, this is supposed to be working.

I'll have to check where the differences are in those 2 flows. By the time they get to the entry point of the tools here, there should be no substantial differences (it's the same function, it should have similar arguments).

deepikabhavnani, 2018-09-13:
@janjongboom - I do not see `LPC1768` as option in `target` . https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-blinky

Not sure if it was removed or we have new error

janjongboom, 2018-09-14:
@deepikabhavnani You need to click *Add to Online Compiler* on the LPC1768 platform page first. https://os.mbed.com/platforms/mbed-LPC1768/

deepikabhavnani, 2018-09-14:
mbed_config.h is not missing for me in zip.
[image]

janjongboom, 2018-09-17:
OK, then I assume it was fixed @theotherjimmy

theotherjimmy, 2018-09-24:
Alright then. Closing.

closed 2018-09-24 by theotherjimmy