ARMmbed/mbed-os #7615: LPC1768 cannot build blinky in uVision5 due to nanostack errors

janjongboom, 2018-07-27:
### Description

1. Export mbed-os-example-blinky (latest) to LPC1768 and uVision5.
1. Patch https://github.com/ARMmbed/mbed-os/issues/7614 manually by adding the file to tree yourself.
1. Build the project.

Result:

```
mbed-os/features/nanostack/sal-stack-nanostack/source/nsconfig.h(48): error:  #5: cannot open source input file "configs/nanostack_full.cfg": No such file or directory
  #include ns_cfg_header(NSCONFIG)
mbed-os/features/nanostack/sal-stack-nanostack/source/Service_Libs/fhss/fhss_beacon_tasklet.c: 0 warnings, 1 error
compiling fhss_beacon.c...
mbed-os/features/nanostack/sal-stack-nanostack/source/nsconfig.h(48): error:  #5: cannot open source input file "configs/nanostack_full.cfg": No such file or directory
  #include ns_cfg_header(NSCONFIG)
mbed-os/features/nanostack/sal-stack-nanostack/source/Service_Libs/fhss/fhss_beacon.c: 0 warnings, 1 error
compiling fhss_channel.c...
mbed-os/features/nanostack/sal-stack-nanostack/source/nsconfig.h(48): error:  #5: cannot open source input file "configs/nanostack_full.cfg": No such file or directory
  #include ns_cfg_header(NSCONFIG)
mbed-os/features/nanostack/sal-stack-nanostack/source/Service_Libs/fhss/fhss_channel.c: 0 warnings, 1 error
compiling fhss_configuration_interface.c...
mbed-os/features/nanostack/sal-stack-nanostack/source/nsconfig.h(48): error:  #5: cannot open source input file "configs/nanostack_full.cfg": No such file or directory
```

I wouldn't know where this cfg is supposed to be, but it's not in my zip file.

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-08-06:
I can reproduce this problem. The file is present in this repository but not present once you export.

These are the configs missing: https://github.com/ARMmbed/mbed-os/tree/master/features/nanostack/sal-stack-nanostack/source/configs

@theotherjimmy The files in question are `.cfg`. Is this supported extension for exporters? 

cc @SeppoTakalo

SeppoTakalo, 2018-08-06:
Why does the exporting drop files from repository?

theotherjimmy, 2018-08-06:
@SeppoTakalo The exporters don't include everything, because that would be _huge_. Instead they only include things that directly affect the build process.

Why are those `.cfg` instead of `.h` (which they appear to be)

0xc0170, 2018-08-08:
We shall this fix for 5.9.5. 

@SeppoTakalo Can we rename the files to supported format (`.h`) ?

SeppoTakalo, 2018-08-08:
@0xc0170 There should not be any major objections against it. It will just break Nanostack builds, so I would prefer to do it in the master repository first.

Reason for `.cfg` file is that it should not be included from normal applications, and therefore we used separate file suffix.

@kjbracey-arm any opinions?

kjbracey-arm, 2018-08-08:
I have no objection to changing them to ".h". Maybe still keep a "cfg" in there somewhere as prefix/suffix just to try to avoid name clashes.

I think the ".cfg" may have originally been a thought about indicating intensely limited syntax to maybe have them parseable by some other non-C-preprocessor tool. They replaced some ".mk" files intended to be included by GNU Make.

0xc0170, 2018-08-10:
> I have no objection to changing them to ".h". Maybe still keep a "cfg" in there somewhere as prefix/suffix just to try to avoid name clashes.

Please send PR with this

artokin, 2018-08-13:
We will create a new PR for the fix, @0xc0170 , would you please merge https://github.com/ARMmbed/mbed-os/pull/7737 as the change is made on top of it.

janjongboom, 2018-08-13:
@0xc0170 Will this still make it into 5.9.5?

cmonr, 2018-08-13:
@janjongboom The 5.9.5 RC was generated last Friday morning (UK time), and since the PR just came up, it will be targeting 5.9.6.

cmonr, 2018-08-13:
@janjongboom https://github.com/ARMmbed/mbed-os/pull/7780#issuecomment-412682510

Thoughts one way or another?

closed 2018-08-21 by cmonr