ARMmbed/mbed-os #7617: NUCLEO-F303K8 not supported by uVision (but has device pack)

janjongboom, 2018-07-27:
### Description

1. Export F303K8 to uVision 5.
1. Message: not supported.
1. uVision 5 has device pack for this target and other nucleo boards are supported, so this is a bug.

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

bcostm, 2018-07-27:
Is it because this board is only mbed2 compliant ? 

I have this error message:
project.py: error: NUCLEO_F303K8 not supported by `uvision5`

Is it the same as Issue https://github.com/ARMmbed/mbed-os/pull/7197 ?

Same error with NUCLEO_F302R8, NUCLEO_F030R8, NUCLEO_F031K6

0xc0170, 2018-07-27:
> Is it because this board is only mbed2 compliant ?

That is a good catch. The error is not accurate (I reproduced it, complains about not supported by uvision) ?

bcostm, 2018-07-27:
It is maybe related to the `default_lib `parameter in the targets.json ?

Because these boards support uARM and ARM but have default_lib = `small`.

janjongboom, 2018-07-30:
So I tested this on 5.9.3, I assume #7197 is already in there?

0xc0170, 2018-07-30:
> So I tested this on 5.9.3, I assume #7197 is already in there?

Yes, it is.

theotherjimmy, 2018-09-24:
So this is resolved then.

closed 2018-09-24 by theotherjimmy