ARMmbed/mbed-os #7618: Micro:bit not supported in uVision 5

janjongboom, 2018-07-27:
### Description

The nRF51 is fully supported, so I'd expect that the micro:bit would be too. Just set it to nrf51822 device pack?

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

yennster, 2018-09-24:
@janjongboom The micro:bit website specifically mentions Keil as a supported debugger with CMSIS-DAP, do you know what more information Mbed needs in order to export from Mbed to Keil? https://tech.microbit.org/software/daplink-interface/#daplink-software

janjongboom, 2018-09-28:
No idea, @theotherjimmy ?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTDEV-1588*

linlingao, 2019-05-30:
@janjongboom I'm going to close this issue since we have submitted a request to the Keil team. You can track the status via the internal Jira link.

closed 2019-05-30 by linlingao