ARMmbed/mbed-os #7626: Utest/Unity pulled in for application code as well

deepikabhavnani, 2018-07-27:
### Description
| mbed-os\features\frameworks\utest\source\utest_shim.o                                                                                   |     92 |     0 |     0 |

Static timeout object "SingletonPtr<Timeout> utest_timeout_object;" is not removed by linker
0x0004c040   0x0000005c   Code   RO         5399    t._ZN12SingletonPtrIN4mbed7TimeoutEE3getEv  BUILD/K64F/ARM/mbed-os/features/frameworks/utest/source/utest_shim.o

Utest/Unity/Greentea code is essentially for mbed test and should not be part of application code. We can add pre-processing macros to skip them from build.

Idea is to emit macros from `mbed test` and use that macro to guard everything related to testing. 

### Issue request type
[ ] Question  
[ ] Enhancement  
[X] Bug  

CC @theotherjimmy

theotherjimmy, 2018-07-27:
We don't need to involve the preprocessor, we could just stop treating projects as if they were tests.

theotherjimmy, 2018-07-30:
@0xc0170 I was under the impression that the code [here](https://github.com/ARMmbed/mbed-os/blob/master/tools/make.py#L210-L215) and [here](https://github.com/ARMmbed/mbed-os/blob/master/tools/make.py#L277) might be colluding to add the testing framework, but it appears that this is not the case. Instead, it's just scanned all of the time (there may have been an `.mbedignore` in `features/frameworks` before).

SeppoTakalo, 2018-07-31:
Pulling the code in is the correct behaviour. Having complex rules to exclude out stuff just makes things harder to maintain.

However, the problem is real, and has been the same case for some of the connectivity driver. Proposed fix is just not right.
On connectivity driver we had to make sure that there would be no C++ objects in global namespace as those would be initialised before `main()` starts, making the initialisation code to call objects constructor means that there is reference to it, and linker cannot drop it.

Various ways exists to fix this. One is to use this `SingletonPtr` class which would return a object only if someone is referring to it. There are other ways as well. For example all network drivers can provide default instance for target, like this: https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Realtek/TARGET_AMEBA/default_wifi_interface.cpp#L22 but that is dropped by linker if nobody refers to it.

So if including test frameworks, or any code, in Mbed OS has RAM/ROM impact when not used, it needs to be fixed properly, not by modifying the build system, but by fixing the references.

p.s. Bear in mind that in case there are templated classes that share code, the file that first refers to it might be referred as the one that "includes" the code. But there are other users. And I suspect that this is the case here. If you remove the file in question completely, the `SingletonPtr` is still included in the final link, but now shown under different file and final RAM/ROM numbers do not change. We did see this same behaviour when fixing connectivity drivers. So before acting, study whether this issue is valid at all.

theotherjimmy, 2018-07-31:
> but we still need this change for reasons below:

I think need is a strong word here.

> we end up with linker error because of 2 main() functions,

There is more than one way to resolve this issue. We could also redirect main for example.

>  While writing OS code, we sometimes need to insert callback / API's just to help in testing which can be easily removed from final application.

This concerns me, as it would prevent us from testing exactly the same code as we say is production ready.

> Saves build time, while compiling application code.

Yeah, it would be better to use a feature/`TESTS/COMMON` dir for this.

0xc0170, 2018-07-31:
> Pulling the code in is the correct behaviour. Having complex rules to exclude out stuff just makes things harder to maintain.

+1

For me, this issue comes to: What is the difference between `mbed compile` vs `mbed test` ?  As some points above indicate, this new macro `MBED_TEST` can lead to usage that we would not encourage. 

1. Is `MBED_TEST` useful? Yes/No
2. If 1. is Yes, how it would be useful and should be used (use cases) ?

deepikabhavnani, 2018-07-31:
> This concerns me, as it would prevent us from testing exactly the same code as we say is production ready.

My best example for this would be testing UART. Add intermediate buffer to test data format or loopback which is required only for testing as it not required for end application to work properly.

SeppoTakalo, 2018-07-31:
For testing uart, the test code may as well call HAL APIs directly.
I really don't see point of selectively building some code in, regarding some macros to be present.

For example, we are perfectly capable of testing Ethernet MAC drivers by linking the test code with the driver, without IP stack.

Effectively, changing the code based on build time macro, how can you actually proof that the production code is working, if test code is not the same? I'm strongly against changing drivers for testing purposes. If interfaces are done properly, test code can utilise any API level.

closed 2018-08-21 by deepikabhavnani