ARMmbed/mbed-os #7633: NRF52832/NRF52: Addressing Errata 89 and 108 on the latest Rev Chip of NRF529832, to fix high current draw.

yogeshk19, 2018-07-27:
There are two errata's that seem to be potentially affecting our current firmware application on NRF52_DK when operating as a BLE central device, which could potentially causes a higher current draw. 
    - Nordic NRF52_DK
    - GCC 
    - mbed-cli
    - MBED OS 5.5.4

We use I2C/TWI to communicate between two different MCU's, one of the MCU being the NRF52832 . Its possible because of Errata 89, we are seeing a consistent current draw of 500 uA after the i2C transaction is started. 

Here is the link to the two errata's and they have mentioned workarounds, however I am not sure how to implement this fix into the MBED-OS 5.5.4 branch, even if I can have this fix as a private build in the mbed-cli build environment.

[NRF52832 Errata 108](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52832.Rev2.errata%2Fdita%2Ferrata%2FnRF52832%2FRev2%2Flatest%2Fanomaly_832_108.html&cp=2_1_1_0_1_30)

[NRF52832 Errata 89](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52832.Rev2.errata%2Fdita%2Ferrata%2FnRF52832%2FRev2%2Flatest%2Fanomaly_832_108.html&cp=2_1_1_0_1_30)

[ ] Question  
[X] Enhancement  
[X] Bug  

Thanks,
Yogesh

yogeshk19, 2018-07-27:
I just wanted to add an update, I found that errata 108 was implemented in 5.9 mbed-os branch and so I just copied the errata 108 related changes from the system_nrf52.c file in mbed os 5.9 branch to mbed os 5.5.4 brach system_nrf52.c file. However If I could get some some guidance wrt implementing the errata 89 that would be great.

Thanks,
Yogesh

yogeshk19, 2018-07-30:
we are still verifying if errata 89 is the root cause of the 500 uA power draw in BLE central. But in our preliminary tests it doesn't seem like it. Anyway why would I require to update the nrf_spim_disable(), isn't that related to the SPI interface? However the errata is more related to I2C interface I think based on the workaround for the errata.

Thanks,
Yogesh

yogeshk19, 2018-07-30:
@TacoGrandeTX Thanks for taking the time to respond to my query. The mbed os 5.5.4 builds against sdk 11, so I added the code in nrf_twi.h, as only changes to this file made i2c_api.c recompile and changes to sdk13 nrf_twim.h doesn't cause a recompile of i2c_api.c. Here is the code I added based on the errata for just the i2c part. I was wondering if you can let me know if I am in the right path.

```
__STATIC_INLINE void nrf_twi_disable(NRF_TWI_Type * p_twi)
{
        p_twi->ENABLE = (TWI_ENABLE_ENABLE_Disabled << TWI_ENABLE_ENABLE_Pos);

        if(p_twi == NRF_TWI0) 
	{
		*(volatile uint32_t *)0x40003FFC = 0;
		*(volatile uint32_t *)0x40003FFC;
		*(volatile uint32_t *)0x40003FFC = 1;
	}
	else if(p_twi == NRF_TWI1) 
	{
		*(volatile uint32_t *)0x40004FFC = 0;
		*(volatile uint32_t *)0x40004FFC;
		*(volatile uint32_t *)0x40004FFC = 1;
	}
}
```

if the above code changes seem right, I will add it for the spi as well and see if adding this change doesn't break i2c in any way and the power draw seems reasonable.

Thanks,
Yogesh

TacoGrandeTX, 2018-07-31:
@Yogeshk19 Sorry that we thought you were using SDK v13.  I personally haven't used SDK v11 but I pulled it down and reviewed it.  For SDK v11 it appears that we don't have EasyDMA support for the TWI (this is called TWIM) even though it is available in the Nordic SDK.  For instance, I don't see the Nordic driver `nrf_drv_twi.c` being delivered in Mbed for SDK v11.  

Are you sure you are using the EasyDMA with the TWI?  If you aren't, then you are not impacted by the errata for the TWIM.  We do have EasyDMA support for the SPI (called SPIM) and your approach to the work around is correct and that could be used in `nrf_spim.h`.