ARMmbed/mbed-os #7651: Help / problem with "extra_labels_remove"

Alex-EEE, 2018-07-30:
### Description
I'm trying to add support for a close part cousin of the mbed supported MCU_K22F512.  I'm having a hard time getting CLI to ignore the mbed K22 linker script and use mine.  I'm trying this:

"AI_K22": {
        "inherits": [ "MCU_K22F512" ],
        "extra_labels_remove": [ "FRDM", "MCU_K22F512" ],
        "extra_labels_add": [ "MCU_K22F512AVLK" ],

But I get this warning:

Scan: ..
Warning: Multiple linker scripts detected: ../mbed/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K22F/TARGET_MCU_K22F512/device/TOOLCHAIN_IAR/MK22F51212.icf -> ../mbed/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K22F/TARGET_MCU_K22F512AVLK/device/TOOLCHAIN_IAR/MK22F51212.icf

How does "extra_labels_remove" work?  I would think that removing "MCU_K22F512" would cause the CLI to not see 
TARGET_MCU_K22F512/device/TOOLCHAIN_IAR/MK22F51212.icf

I want it to use 
TARGET_MCU_K22F512AVLK/device/TOOLCHAIN_IAR/MK22F51212.icf

Am I doing this the right way?

### Issue request type

[ x ] Question  
[ ] Enhancement  
[ ] Bug

Alex-EEE, 2018-07-31:
Yea, exactly, my part is actually just a custom board and a very similar but still different K22.  I need to have IAR export use a different startup and linker script, and to also use different PinDef files.

Alex-EEE, 2018-08-10:
@0xc0170 ?

0xc0170, 2018-08-10:
Extra label remove should work. I can see that the target is named and extra label added (this is not needed).  Anything changes if you remove first manually the duplication in extra labels ?

However as target is named MCU_ (this one you want to remove), not certain if this is supported

@theotherjimmy Please review

theotherjimmy, 2018-09-24:
@Alex-EEE You cannot remove target names from the labels. This is working as designed. You may have to add an `.mbedignore` for those few files.

closed 2018-09-24 by theotherjimmy

Alex-EEE, 2018-09-24:
Ok, so you're basically telling me that mbed-os cannot simultaneously support two different variations of the MK22 chip?  B/c as I've pointed out, I need to get the toolchain to not use the mbed provided MK22 files, but rather, it needs to use the other MK22 folder for a custom target.

Using .mbedignore would remove the folder I don't want for ALL TARGETS.

How do I remove folders for only ONE TARGET.

Why does extra_label_remove not work for targets?  This doesn't make sense.  Allowing the removal of this other target folder would allow my custom MK22 target to compile correctly.

@theotherjimmy am I missing something here?  How am I supposed to support two variants of MK22 with this current restriction in the CLI?