ARMmbed/mbed-os #7656: test: Automated Singletest MBED_24 is failing with all supported tool-chains for Toshiba's TMPM066.

ganesh-ramachandran, 2018-07-31:
### Description

**Target:** 
Toshiba's TMPM066

**M066 Memory:** 
RAM - 16K and Flash - 128K

**Tool Chain:**
GCC_ARM / ARM / IAR

**Issue:**
- Existing implementation of us_ticker is with timerA. After adding extended feature, some greentea testcases were failing. 
- To solve this, ticker file is newly implemented with timerB.
- With the new us_ticker file all greentea cases are passed but facing the issue with Singletest automated testcase _MBED_24_ ("Timeout Int us”) with all supported tool-chains.

**Expected behavior**
Test case should complete in 0.90 Sec to 1.10 Sec.

**Actual behavior**
Test case is completing in 1.23 Sec

**Steps to reproduce**
1. Download master from https://github.com/ganesh-ramachandran/mbed-os (since us_ticker file is newly implemented) or
   From https://github.com/ganesh-ramachandran/mbed-os, download the newly implemented mbed-os\targets\TARGET_TOSHIBA\TARGET_TMPM066\us_ticker.c and targets.json 
2. Build command
```
python tools/singletest.py --auto --tc GCC_ARM -n MBED_24 -v
```

**Observation:**
   - "attach_us" function is taking up to 200us to process at 24Mhz system clock.
   - This function is called 2000 times in the test case. This additional delay resulting in failure of test case.
   - With Ticker object "attach_us" function is called only once.

If we use Ticker object in test case instead of Timeout object, observed that the Test case is Passed.

**Note:**
Maximum Operating frequency for TMPM066 is 24 MHz

**Log file:**
[MBED_24_Test_Log.txt](https://github.com/ARMmbed/mbed-os/files/2244536/MBED_24_Test_Log.txt)

[ ] Question  
[ ] Enhancement  
[x] Bug

ganesh-ramachandran, 2018-07-31:
@0xc0170 Thanks for the update.
Shall we close this issue?

0xc0170, 2018-07-31:
I would like this to be reviewed by core team. Once it is mirrored, we will triage it

jamesbeyond, 2018-08-14:
Hi @ganesh-ramachandran,
Sorry for the late reply.

To clarify the thing:
The scripts you used and tests run are for the legacy mbed os 2 tests. and they had been announced deprecated. We have plans to remove them soon.

For your target: `TMPM066` it clearly declared that it only support mbed 5 in  [here](https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L4193)

To test your targets with mbed OS 5 with the built in tests please following the testing section in the mbed os [docs](https://os.mbed.com/docs/v5.9/tools/testing-applications.html)

Or simply just run:
```
mbed test -t GCC_ARM -m TMPM066
```

jamesbeyond, 2018-08-14:
@ganesh-ramachandran  If you are happy about the solution, I think this issue can be closed

closed 2018-08-16 by ganesh-ramachandran